united-manufacturing-hub/united-manufacturing-hub #223: Explaining timescaleDB vs influxDB

JeremyTheocharis, 2021-05-03:
Explain why we chose timescaleDB over influxDB.

closed 2021-05-07 by JeremyTheocharis

valyala, 2021-11-04:
Did you evaluate other time series databases for storing measurements data? For example, [VictoriaMetrics](https://github.com/VictoriaMetrics/VictoriaMetrics) (I'm the core developer of this TSDB). It has the following features:

* It provides better query language than TimescaleDB for typical time series data - [MetricsQL](https://docs.victoriametrics.com/MetricsQL.html), which is backwards-compatible with PromQL - query language for [Prometheus](https://prometheus.io). For example, the following SQL query mentioned at https://github.com/united-manufacturing-hub/united-manufacturing-hub/commit/da1844c99c8e67fabe173d6f9f405c8d16406eda :
```sql
SELECT time, (memUsed / procTotal / 1000000) as value
FROM measurements
WHERE time > now() - '1 hour';
```
can be re-written into the following MetricsQL query:
```metricsql
memUsed / procTotal / 1e6
```
Read more details about MetricsQL and PromQL in [this tutorial](https://valyala.medium.com/promql-tutorial-for-beginners-9ab455142085).

* It accepts data in [many popular data ingestion formats](https://docs.victoriametrics.com/#how-to-import-time-series-data) including InfluxDB, Graphite, Prometheus, OpenTSDB, DataDog, CSV and JSON.

* It usually provides many times better on-disk data compression than TimescaleDB even when the compression is [enabled in TimescaleDB](https://docs.timescale.com/timescaledb/latest/how-to-guides/compression/). See [this article for details](https://abiosgaming.com/press/high-cardinality-aggregations/).

* It uses much lower amounts of disk IO than TimescaleDB, so it can run in resource-constrained IoT environments with slow storage. See [this benchmark](https://valyala.medium.com/promscale-vs-victoriametrics-resource-usage-on-production-workload-91c8e3786c03) for details.

JeremyTheocharis, 2021-11-07:
This sounds interesting. You could open a discussion for this, so your idea does not get lost in an old issue :)

commit da1844c9 (2021-05-07): 223 timescaledb vs influxdb (#237)

Fixes #223