united-manufacturing-hub/united-manufacturing-hub #239: Update architecture on getting started

kruegeralex, 2021-05-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

alphabet5, 2021-05-12:
If the architecture drawing could incorporate IDMZ requirements would be excellent. 

Current architecture drawing appears to bypass 1 of the following design principles:
- All IACS network traffic from either side of the IDMZ terminates in the IDMZ; no IACS traffic directly traverses the IDMZ
- IACS traffic does not enter the IDMZ; it remains within the Industrial Zone

I would imagine the broker could be placed in the IDMZ, or proxy the MQTT traffic with haproxy, IOXY, etc. I haven't tried proxying mqtt before though, and I'm not familiar enough with the components you are using.

JeremyTheocharis, 2021-05-13:
Thank your for the feedback! 

This is exactly what we recommend when connecting directly with the PLC, e.g. with our certified device https://docs.umh.app/docs/tutorials/certified-devices/machineconnect/

There will be a firewall in front of the internal stack and the PLC, which not only prevents external traffic from outside, but is also able to log and throw alarms if suspicious traffic is detected (IDS functionality from OPNsense).

By default, we allow the PLC to communicate with the outside world, however, that is easily adjusted. To communicate with the PLC you have to be either physically at the machine, connected with VPN or use MQTT.

Is that what you are recommending?

alphabet5, 2021-05-13:
The specific section stating to allow communication to the PLC would be breaking IDMZ design principles.

Here's a high-level overview showing the communication.

[image]

I would also caution about recommending DHCP in the manufacturing zone.

"Cisco and Rockwell Automation recommend that IACS network developers use a static IP addressing schema for the Manufacturing zone, especially for allocating IP addresses to IACS devices in the Cell/Area zone. Cisco and Rockwell Automation now recommend DHCP Persistence as a valid option along with static addressing for deploying IP addresses for IACS devices."

JeremyTheocharis, 2021-05-13:
I think we are actually on the same page regarding security, but I am bad at explaining. The edge device itself is a DMZ as it consists out of multiple virtual machines with a firewall.

[image]

I also created a separate issue for that: #245

Or is there still something violating the DMZ approach? We are always happy to get feedback and/or explain more!

Also regarding DHCP: we recommend giving the PLC a static address via DHCP, which is the same word as "DHCP persistence" (Configure OPNsense to give out the same IP for the MAC-address of the PLC for LAN. Go to Services –> DHCPv4 –> LAN and add the PLC under “DHCP static mappings for this device”)

closed 2021-06-09 by JeremyTheocharis