united-manufacturing-hub/united-manufacturing-hub #244: Support for Ethernet/IP (CIP) PLCs

alphabet5, 2021-05-12:
**Is your feature request related to a problem? Please describe.**
Support for Allen Bradley PLC's using Ethernet/IP. Most of the PLC's I deal with (in North America) are AB ControlLogix or CompactLogix. Support for these PLC's without having to convert from Ethernet/IP to OPCUA (using ignition, kepware, ftlinx, etc.) would be idea.

**Describe the solution you'd like**

**Describe alternatives you've considered**
kepware/ignition/ftlinx for enip->opcua seems like it would be the alternative. Built-in support would be excellent. 

**Additional context**
[libplctag](https://github.com/libplctag/libplctag) could be used. They provide an example wrapper in go that could be used, and the licensing would be compatible as far as I can tell.

JeremyTheocharis, 2021-05-13:
Hi, thank you for the idea!

For connecting machines we are typically using node-red, as it is open source and has a lot of community built interfaces (like OPC/UA). There is also a node available that looks exactly like what you need: https://flows.nodered.org/node/node-red-contrib-cip-ethernet-ip

Have you tried that? As we do not have a PLC on hand we are currently unavailable to test it ourselves and therefore it is not included by default.

alphabet5, 2021-05-13:
I will give it a try. I missed the part where node red is being used for the plc communication. Typically this is avoided due to performance reasons. 

For a typical site we are reading ~5-20k tags/second from a half dozen PLC's to historize the values, along . Unoptimized instances of polling these PLC's will, and have pegged the CPU leading to downtime. 

I'll see if I can get a test instance setup and do some performance testing. Thanks!

JeremyTheocharis, 2021-05-13:
Fully agreed! node-red or Javascript in general is not designed for performance. For everything that we routinely do in node-red we write a entire seperate microservice like sensorconnect, which do it way more performant.

JeremyTheocharis, 2021-07-01:
@alphabet5 Any feedback?

@kruegeralex Didn't our customer in brazil successfully use Ethernet IP using the node-red flow? If yes, we should add that to the documentation and then we can close this issue.

alphabet5, 2021-07-02:
I haven't had a chance to do any testing yet. If you have documentation for it that would be great.

kruegeralex, 2021-07-16:
Our customer had not used ethernet-ip but used: https://flows.nodered.org/node/node-red-contrib-pccc
to read out his Rockwell PLC. This worked without any problems

Scarjit, 2023-03-31:
Should be fixed by integrating Benthos.
Please re-open if not.

closed 2023-03-31 by Scarjit