united-manufacturing-hub/united-manufacturing-hub #256: Connect to locally hosted Kubernetes Stack (MQTT-Broker) from locally hosted Docker Container

JonaVo, 2021-05-27:
I would like to connect the MQTT-Broker included in your stack from my own docker container, running locally on my system. Currently, I cannot reach the MQTT-Broker.
I appreciate your help.

JonaVo, 2021-05-27:
I managed to fix it :)
If you are facing the same problem, this might help you:
I created an additional docker network with the docker host and my local docker. This allows you to directly address the docker host from inside your docker using the IP: 192.168.0.1.

Here is my docker-compose file:

        version: '3.7'
        
        services:
          camera_connect:
            image: camera_connect_image
            build:
                context: ../
                dockerfile: ./docker/Dockerfile
            container_name: camera_connect_container
            env_file:
              - ./.env
            volumes:
              - "./images:/app/images/"
            networks:
              - camera_vlan
              - dockernet
        
        networks:
          camera_vlan:
            driver: macvlan
            driver_opts:
              parent: enp1s0 #adjust if installed on different system
            ipam:
              driver: default
              config:
                - subnet: "192.168.1.0/24"
          dockernet:
            driver: bridge
            ipam:
              driver: default
              config:
                - subnet: "192.168.0.0/24"

JonaVo, 2021-05-27:
Port is 1883 by the way

JeremyTheocharis, 2021-05-30:
If you want to connect to the MQTT Broker from inside the local Kubernetes cluster, you can use the following URI `mqtt://factorycube-edge-emqxedge-headless:1883`. 

if you want to connect from outside, e.g. a locally running Docker container or from externally, you can replace `factorycube-edge-emqxedge-headless` with the IP of the edge device, as the Kubernetes cluster will automatically forward the port 1883 to be available externally.

In the case of @JonaVo the edge device has the IP 192.168.0.1, therefore connecting with 192.168.0.1 on port 1883 will work.

closed 2021-05-30 by JeremyTheocharis