united-manufacturing-hub/united-manufacturing-hub #289: IFM Sensor O1D108 is not processed automatically

kruegeralex, 2021-06-30:
Provide a general summary of the issue in the Title above

## Expected Behavior

when i connect this sensor i want to get automatically the interpreted values (distance etc.)

## Current Behavior

Tell us what happens instead of the expected behavior

The service returns the values only as value_string

Scarjit, 2022-06-03:
@kruegeralex Is this still an issue with sensorconnect ?

Scarjit, 2023-03-31:
Is this still relevant @kruegeralex ?

EmlingerM, 2024-10-11:
I could reproduce this issue. The problem is that all values, regardless of their actual type, are being returned in the JSON payload as strings.

```json
{
   "Distance":"47",
   "Switch state [OUT1]":"1",
   "connected":"connected",
   "timestamp_ms":"1728629682975",
   "type":"Io-Link"
}
```

Regarding https://umh.docs.umh.app/docs/datamodel/messages/_historian/#message-structure, the payload should look like this:

```json
{
   "Distance": 47,
   "Switch state [OUT1]": true,
   "connected":"connected",
   "timestamp_ms": 1728629682975,
   "type":"Io-Link"
}
```

closed 2024-10-11 by Scarjit