united-manufacturing-hub/united-manufacturing-hub #306: AL1353 keeps crashing uncontrollably

JeremyTheocharis, 2021-07-13:
## Issue

By default sensorconnect scans the network for devices and reads them out with the highest frequency. For one of our customers sensorconnect suddenly stops reading out this device, so that no data appears in the MQTT broker.

## Data

After a restart of sensorconnect (killing the pod) it works again and shows these logs:

```
Tue Jul 13 12:25:44 2021 Searching for new devices
Tue Jul 13 12:25:45 2021 MQTT Connected
Tue Jul 13 12:25:46 2021 Unexpected disconnection.
Tue Jul 13 12:25:47 2021 MQTT Connected
Tue Jul 13 12:25:48 2021 Unexpected disconnection.
Tue Jul 13 12:25:49 2021 MQTT Connected
Tue Jul 13 12:25:50 2021 Unexpected disconnection.
Tue Jul 13 12:25:51 2021 MQTT Connected
Tue Jul 13 12:25:52 2021 Unexpected disconnection.
Tue Jul 13 12:25:53 2021 MQTT Connected
Tue Jul 13 12:25:54 2021 Unexpected disconnection.
Tue Jul 13 12:25:55 2021 MQTT Connected
Tue Jul 13 12:25:56 2021 Unexpected disconnection.
Tue Jul 13 12:25:57 2021 MQTT Connected
Tue Jul 13 12:25:58 2021 Unexpected disconnection.
Tue Jul 13 12:25:59 2021 MQTT Connected
Tue Jul 13 12:26:00 2021 Unexpected disconnection.
Tue Jul 13 12:26:01 2021 MQTT Connected
Tue Jul 13 12:26:11 2021 1 devices discovered. 1 of them are new.
Tue Jul 13 12:26:11 2021 Starting thread for: 192.168.10.204 | AL1353 | REDACTED
Tue Jul 13 12:26:11 2021 Waiting 10 seconds till next discover
Tue Jul 13 12:26:21 2021 Searching for new devices
Tue Jul 13 12:26:49 2021 0 devices discovered. 0 of them are new.
Tue Jul 13 12:26:49 2021 Waiting 10 seconds till next discover
```

The gateway shows the following screen:
[image]

When the AL1353 crashes the next time we will upload the entire log of sensorconnect.

## Ideas

- MQTT disconnection (however, emqxedge is running continuously)
- Weird bug in sensorconnect (is unlikely)
- network cable issue

## Next steps
- add continuous ping in Node-RED
- check material (specifically network cables)
- get full logs of sensorconnect when the connection crashes the next time

kruegeralex, 2021-07-15:
uploading the entire log of sensorconnect:
[factorycube-edge-sensorconnect-bc46f9966-8mklx.log](https://github.com/united-manufacturing-hub/united-manufacturing-hub/files/6823490/factorycube-edge-sensorconnect-bc46f9966-8mklx.log)

Scarjit, 2022-03-03:
Fixed with new sensorconnect

closed 2022-03-03 by Scarjit