united-manufacturing-hub/united-manufacturing-hub #309: Add productTag to MQTT

larswitte, 2021-07-13:
Regarding issue: #297

**Problem:**
MQTT topic necessary to send contextualized process information. Right now it is not possible to send processValues tied to a single uid.

**Possible Solution**
Add productTag MQTT Topic to UMH datamodel and add table productTagTable to TimescaleDB database.

Topic: ia/customerID/location/AssetID/productTag

**Example 1**
{
         "timestamp_ms": 215348452385
         "uid": 14432504350
         "productID": "test123"
         "name": "barcode"
         "value": "234090adflkj34"
         "isScap": false
}

**Example 2**
{
        "timestamp_ms": 1243204549
        "uid": 32493855304
        "productID": "test324"
        "name": "torque1"
        "value": 1234   // problem with different types?
        "isScrap":  false
}

JeremyTheocharis, 2021-07-14:
Let's take isScrap out and put it into a seperate MQTT message called `/scrapUniqueProduct`

larswitte, 2021-07-15:
Good idea, thanks! 

**Example ("isScrap" and "productID" removed)**
Topic: `ia/customerID/location/AssetID/productTag`
```js
{
         "timestamp_ms": 215348452385
         "uid": 14432504350
         "name": "barcode"
         "value": 2450987
}
```
We will also differentiate between `/productTag` (float) and `/productTagString` (string). I opened a new issue #319 for  `/productTagString`.

larswitte, 2021-07-21:
Changed product Tag to:
``` .js
{
         "timestamp_ms": 215348452385
         "aid": "14432504350"
         "name": "barcode"
         "value": 2450987
}
```

closed 2021-09-23 by larswitte