united-manufacturing-hub/united-manufacturing-hub #313: Factoryinsight: Add REST API's for productTag data from TimescaleDB

larswitte, 2021-07-14:
**Description**
To access the productTagTable Data (used for digital shadow, track and trace: #297) we need to provide new REST API's.

**New REST API's**
- get /{customer}/{location}/{asset}/productTags
from ```<timestamp1>``` to ```<timestamp2>``` (in RFC 3999 Format)
Example: (Two data points. The data for one single part at one asset is combined together to one data point.)
```
{
	"columnNames": [
		"ValveHeadID",
		"TorqueScrew1",
		"TorqueScrew2",
		"TorqueScrew3",
		"TorqueScrew4",
		"GasketImageID",
		"UID"
		"Timestamp_ValveHeadID"
		"Timestamp_GasketImageID"
	]
	"datapoints": [
		[
			"124000003941",
			5.2,
			5.4,
			4.9,
			5.4,
			"345g48t34ru2,
			"2502934850239587"
			1239481702
			2314092348
		]
			"124000003942",
			5.1,
			5.1,
			4.8,
			5.9,
			"2493trf8434r",
			"32ß4529ß8359"
			13204982743
			12359873404
		]
	]
}
```

- get /{customer}/{location}/digitalProductShadow
from ```<timestamp>``` to ```<timestamp>``` (in RFC 3999 Format)
for specific ```<ValveHeadID>``` or ```<HeatExchangerID>``` or ```<CompressorID>``` or ```<EBoxID>```.
Example (with one datapoint):
```
{
	"columnNames": [
		"ValveHeadID",
		"HeatExchangerID",
		"EBoxID"
		"CompressorID",
		"ProductID"
		"Washing_Timestamp",
		"Preassembly_Timestamp",
		"Preassembly_ScrewBatch",
		"Assembly_1_TorqueScrew1",
		"Assembly_1_TorqueScrew2",
		"Assembly_1_TorqueScrew3",
		"Assembly_1_TorqueScrew4",
		"Assembly_1_GasketImageID",
		"Assembly_1_Timestamp_Scan",
		"Assembly_1_Timestamp_Image",
		"Assembly_2_Timestamp"
		"Assembly_3_Timestamp"
	]
	"datapoints": [
		[
			"124000003941",
			"848000001248",
			"256000002348",
			"167000002348",
			"TestProduct123"
			124832048,
			234879273,
			"TestScrewBatch34",
			4.6,
			5.5,
			4.9,
			5.4,
			"345g48t34ru2",
			3249238474
			2349874343
			1239481702
			2314092348
		]
}
```

- get /{customer}/{location}/{asset}/productTagsRaw
from ```<timestamp>``` to ```<timestamp>``` (in RFC 3999 Format)
Maybe we need to split this request in two: one for float values and one for strings.
Example (two Datapoints)
```
{
	"columnNames": [
		"AssetID"
		"timestamp_ms",
		"uid"
		"product_id"
		"valueName"
		"value",
	]
	"datapoints": [
		[
			"Assembly_1",
			12439248,
			"testUID234981734",
			"TestProduct123",
			"barcode",
			"167000002348"
		],
		[
			"Assembly_1",
			124392676,
			"testUID234981737",
			"TestProduct123",
			"Torque1"
			5.4
		]
}
```

JeremyTheocharis, 2021-07-14:
- get /{customer}/productTag
`uniqueProductID`

Maybe we need to split this request in two: one for float values and one for strings.

Example (two Datapoints)
```
{
	"columnNames": [
		"AssetID"
		"timestamp_ms",
		"uid"
		"product_id"
		"valueName"
		"value",
	]
	"datapoints": [
		[
			"Assembly_1",
			12439248,
			"testUID234981734",
			"TestProduct123",
			"barcode",
			"167000002348"
		],
		[
			"Assembly_1",
			124392676,
			"testUID234981737",
			"TestProduct123",
			"Torque1"
			5.4
		]
}
```

This function returns all stored productTags for one uniqueProduct (incl. duplicates)

- get /{customer}/{location}/{asset}/uniqueProductsWithTags
from ```<timestamp1>``` to ```<timestamp2>``` (in RFC 3999 Format)

This function returns all uniqueProducts for that specific asset in the given timerange including all unqiueProductTags (value and timestamp for each uniqueProduct.

What happens if there is no productTag for that uniqueProduct? (probably `null`). What happens if there are multiple values for one uniqueProductTag (error?)

Example: (Two data points. The data for one single part at one asset is combined together to one data point.)
```
{
	"columnNames": [
		"ValveHeadID",
		"TorqueScrew1",
		"TorqueScrew2",
		"TorqueScrew3",
		"TorqueScrew4",
		"GasketImageID",
		"UID"
		"Timestamp_ValveHeadID"
		"Timestamp_GasketImageID"
	]
	"datapoints": [
		[
			"124000003941",
			5.2,
			5.4,
			4.9,
			5.4,
			"345g48t34ru2,
			"2502934850239587"
			1239481702
			2314092348
		]
			"124000003942",
			5.1,
			5.1,
			4.8,
			5.9,
			"2493trf8434r",
			"32ß4529ß8359"
			13204982743
			12359873404
		]
	]
}
```

- get /{customer}/digitalProductShadow
from ```<timestamp>``` to ```<timestamp>``` (in RFC 3999 Format)

for specific ```uniqueProductID``` (uid)

This function get all productTags for all parents and children for a given uniqueProductID.

Example (with one datapoint):
```
{
	"columnNames": [
		"ValveHeadID",
		"HeatExchangerID",
		"EBoxID"
		"CompressorID",
		"ProductID"
		"Washing_Timestamp",
		"Preassembly_Timestamp",
		"Preassembly_ScrewBatch",
		"Assembly_1_TorqueScrew1",
		"Assembly_1_TorqueScrew2",
		"Assembly_1_TorqueScrew3",
		"Assembly_1_TorqueScrew4",
		"Assembly_1_GasketImageID",
		"Assembly_1_Timestamp_Scan",
		"Assembly_1_Timestamp_Image",
		"Assembly_2_Timestamp"
		"Assembly_3_Timestamp"
	]
	"datapoints": [
		[
			"124000003941",
			"848000001248",
			"256000002348",
			"167000002348",
			"TestProduct123"
			124832048,
			234879273,
			"TestScrewBatch34",
			4.6,
			5.5,
			4.9,
			5.4,
			"345g48t34ru2",
			3249238474
			2349874343
			1239481702
			2314092348
		]
}
```

- get /{customer}/digitalProductShadowFromTag
from ```<timestamp>``` to ```<timestamp>``` (in RFC 3999 Format)

for specific `tagName`, `tagValue`

This function get all productTags for all parents and children for a given `tagName`, `tagValue`.

Example (with one datapoint):
```
{
	"columnNames": [
		"ValveHeadID",
		"HeatExchangerID",
		"EBoxID"
		"CompressorID",
		"ProductID"
		"Washing_Timestamp",
		"Preassembly_Timestamp",
		"Preassembly_ScrewBatch",
		"Assembly_1_TorqueScrew1",
		"Assembly_1_TorqueScrew2",
		"Assembly_1_TorqueScrew3",
		"Assembly_1_TorqueScrew4",
		"Assembly_1_GasketImageID",
		"Assembly_1_Timestamp_Scan",
		"Assembly_1_Timestamp_Image",
		"Assembly_2_Timestamp"
		"Assembly_3_Timestamp"
	]
	"datapoints": [
		[
			"124000003941",
			"848000001248",
			"256000002348",
			"167000002348",
			"TestProduct123"
			124832048,
			234879273,
			"TestScrewBatch34",
			4.6,
			5.5,
			4.9,
			5.4,
			"345g48t34ru2",
			3249238474
			2349874343
			1239481702
			2314092348
		]
}
```

- get /{customer}/uniqueProductFromTag
`tagName`, `tagValue`

Example (two Datapoints)
```
{
	"columnNames": [
		"uid"
		"valueName"
		"value",
	]
	"datapoints": [
		[
			"testUID234981734",
			"barcode",
			"167000002348"
		],
		[
			"testUID234981735",
			"barcode",
			"167000002348"
		],
}
```

JeremyTheocharis, 2021-07-20:
@larswitte I stopped working on it as I am unsure whether we might want to use `uniqueProductAlternativeID` instead of using `tagName` and `tagValue`

closed 2021-09-23 by larswitte