united-manufacturing-hub/united-manufacturing-hub #330: Remove Grafana datasources

JeremyTheocharis, 2021-07-20:
We should remove the Grafana datasources once we have moved everything to https://github.com/united-manufacturing-hub/united-manufacturing-hub-datasource

JeremyTheocharis, 2021-07-22:
Additionally, we need to automatically update `deployment/factorycube-server/grafanaDatasource/plugin.zip`

Scarjit, 2021-08-23:
> Additionally, we need to automatically update `deployment/factorycube-server/grafanaDatasource/plugin.zip`

Do we really need to update that ?
Our values.yaml in factorycube-server automatically pulls the latest release of both plugins:
https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/main/deployment/factorycube-server/values.yaml#L1006

JeremyTheocharis, 2021-08-24:
You are right!

Scarjit, 2021-08-30:
The only remaining traces of Grafana inside this repo are grafana-proxy, documentation & helm charts.
Therefore i don't see any remaining work for this issue.

closed 2021-08-30 by Scarjit