united-manufacturing-hub/united-manufacturing-hub #336: Rest API: Single most important API, sends asset specific data

larswitte, 2021-07-21:
**Functionality**
This function returns all uniqueProducts for that specific asset in the given timerange. One Datapoint contains one childUID and all parentUID's. All unqiueProductTags and uniqueProductTagStrings (value and timestamp for each uniqueProduct) for the childUID are returned in the same Datapoint.

`get /{customer}/{location}/{asset}/uniqueProductsWithTags`
from `<timestamp1>` to `<timestamp2>` (in RFC 3999 Format)

**Open Questions**
- What happens if there is no productTag for that uniqueProduct? (probably null). 
- What happens if there are multiple values for one uniqueProductTag (error?)

**Example**
(Two data points. The data for one single part at one asset is combined together to one data point.)
```.js
{
	"columnNames": [
		"ValveHeadID",
		"TorqueScrew1",
		"TorqueScrew2",
		"TorqueScrew3",
		"TorqueScrew4",
		"gasketUID",
		"childUID"
		"Timestamp_ValveHeadID"
		"Timestamp_GasketImageID"
	]
	"datapoints": [
		[
			"124000003941",
			5.2,
			5.4,
			4.9,
			5.4,
			"345g48t34ru2,
			"2502934850239587"
			1239481702
			2314092348
		]
			"124000003942",
			5.1,
			5.1,
			4.8,
			5.9,
			"2493trf8434r",
			"32ß4529ß8359"
			13204982743
			12359873404
		]
	]
}
```

This issue was split out of the #313 issues and is for the digital shadow issue in general #297.

Gornoka, 2021-07-21:
Format:
``` json
[
   row

]

row:
{
"ValveHeadAlternativeUniqueID": str<123>
"TorqueScrew1": number<5.0>
"TorqueScrew2": number<5.0>
"TorqueScrew3": number<5.0>
"TorqueScrew4": number<5.0>
"gasketUID": number<5.0>
"childUID": number<5.0>
"Timestamp_ValveHeadID": number<5.0>
"Timestamp_GasketImageID": number<5.0>
}
	
´´´

example
´´´json
[
{
"ValveHeadAlternativeUniqueID": str<123>
"TorqueScrew1": number<5.0>
"TorqueScrew2": number<5.0>
"TorqueScrew3": number<5.0>
"TorqueScrew4": number<5.0>
"gasketUID": number<5.0>
"childUID": number<5.0>
"Timestamp_ValveHeadID": number<5.0>
"Timestamp_GasketImageID": number<5.0>
},
{
"ValveHeadAlternativeUniqueID": str<124>
"TorqueScrew1": number<5.0>
"TorqueScrew2": number<5.0>
"TorqueScrew3": number<5.0>
"TorqueScrew4": number<5.0>
"gasketUID": number<5.0>
"childUID": number<5.0>
"Timestamp_ValveHeadID": number<5.0>
"Timestamp_GasketImageID": number<5.0>
},
{
"ValveHeadAlternativeUniqueID": str<125>
"TorqueScrew1": number<5.0>
"TorqueScrew2": number<5.0>
"TorqueScrew3": number<5.0>
"TorqueScrew4": number<5.0>
"gasketUID": number<5.0>
"childUID": number<5.0>
"Timestamp_ValveHeadID": number<5.0>
"Timestamp_GasketImageID": number<5.0>
},
]

JeremyTheocharis, 2021-07-21:
@Gornoka Something went wrong while formatting your mesage. Can you check?

larswitte, 2021-07-21:
Thank you @Gornoka!

I was also thinking about how to implement the functionality (how to get the data from timescaleDB):
1. Get all productUID's from uniqueProductTable within the specified time and from the specified asset.
2. Check if UID's are listed as a child in the productInheritanceTable and select the UID's who are.
3. Get all key, value pairs from the productTagTable, productTagStringTable and all parentUID's from the productInheritanceTable for each of the selected UID's
4. Get the AID's for the parentUID's
5. Return all parent AID's, the child UID and AID, the productTag and the productTagString values in the way @Gornoka proposed.

**Important**
Not all uniqueProducts from one asset are going to be displayed. Only the ones actually used for the output product of the station (child) are returned). Otherwise we would have a unspecified number of scrapped parts included.

Gornoka, 2021-07-21:
> @Gornoka Something went wrong while formatting your mesage. Can you check?

The data is not valid JSON, but the code window is formatted in JSON, that is intentional, the definition has to be done in openapi or something similar.

larswitte, 2021-07-22:
After thinking about it we probably need to specify the location further.
In the documentation: https://docs.umh.app/docs/concepts/mqtt/  we find following sentence `An asset can be a machine, plant or line (Explicitly not a single station of an assembly cell).`  If we now have multiple different EPCIS steps in one asset we get a problem (-> either only one epcis event is easily accessible or we get multiple EPCIS events mixed up back from the API, which is not good style and would cause problems later).

To seperate between different epcis events performed at one asset I think the best option would be to additionally specify the station_id field from the uniqueProductTable for a request. That way we can make shure that we only get one specific EPCIS event processes back. This would increase overall clarity.

That being said, most times probably we just have one station_id per asset (depending on the size definition of asset).

**This would change**
1. Get all productUID's from uniqueProductTable within the specified time and from the specified asset **and station**.

Even better (performance wise) would be using a unique station_id so we don't have to seperately specify the assetID:
1. Get all productUID's from uniqueProductTable within the specified time and from the specified **station**.

> Thank you @Gornoka!
> 
> I was just thinking about how to implement the functionality (how to get the data from timescaleDB, look up planned structure in issue #297):
> 
> 1. Get all productUID's from uniqueProductTable within the specified time and from the specified asset.
> 2. Check if UID's are listed as a child in the productInheritanceTable and select the UID's who are [now only childs in the specified time and at the specified asset are selected]. (I don't think we need to have the Asset stored in the productInheritanceTable because we dont need to filter for the asset at this step. Parents who got a UID at the Asset (because not seen before at another station) are not selectet at step 2. because there is now way they become a child)
> 3. Get all parentUID's from the productInheritanceTable for each of the selected UID's
> 4. Get the AID's for the parentUID's
> 5. Get all key, value pairs from the productTagTable and productTagStringTable
> 6. Return all parent AID's, the child UID and AID, all the productTag and all the productTagString values in the way @Gornoka proposed.
> 
> **Important**
> Not all uniqueProducts from one asset are going to be displayed. Only the ones actually used for the output product (child) of the station are returned. Otherwise we would have a unspecified number of scrapped parts included.

larswitte, 2021-07-23:
We can also drop step 2. because we only create child uniqueProducts. If we have to create a not yet seen uniqueProduct at one station, we would set the asset to a placeholder (e.g. storage). Only the child would get the actual asset assigned.
-> step 1. is enough

PR #465 (2021-08-25): Added working openapi specification
Fixes #336
commit f982e0bd (2021-08-25): Added working openapi specification (#465)

* Added working openapi specification

Fixes #336

* added review items to branch (#466)

* added generation script for factory insight

minor changes in factoryinsight

* added generator scripts for linux
commit 2354e0b8 (2021-09-01): 336-api-digital-shadow (#461)

Fixes #336 

* initial commit: factoryinsight changes for ds

* worked on sql request for api

* corrected small variable mistake

* worked on the sql query and pivoting

* writing tag data in [][]interface{}

* string tag processing

* worked on productTagString part of request

* dataprocessing

* work on sql query

* first review of database.go done

* used newColumns Map to check for not yet added productTags

* second review of database.go done

* changed zap Debug to zap Errorf messages

* fixed for loop to add new column

* added case: no row in data.Datap [... cut]
commit 4be58a82 (2021-09-09): feat(factoryinsight): added digital shadow (#461)

Fixes #336

* initial commit: factoryinsight changes for ds

* worked on sql request for api

* corrected small variable mistake

* worked on the sql query and pivoting

* writing tag data in [][]interface{}

* string tag processing

* worked on productTagString part of request

* dataprocessing

* work on sql query

* first review of database.go done

* used newColumns Map to check for not yet added productTags

* second review of database.go done

* changed zap Debug to zap Errorf messages

* fixed for loop to add new column

* added case: n [... cut]
closed 2021-09-09 by JeremyTheocharis