united-manufacturing-hub/united-manufacturing-hub #350: SQL Database to connect Tableau to

larswitte, 2021-07-22:
**Overview**
For the digital shadow functionality (issue #297) we need to give the tableau server access to the data. Because the tableau server can't directly connect to the REST API (issue #336) we need to either use a database in between or a web data connector. I was advised against the tableau web data connector (general info: https://help.tableau.com/current/pro/desktop/en-us/examples_web_data_connector.htm) so the plan is to go with a sql database.

**Requirements**
- very compatible with node red (microservice connects to the REST API  (issue #336) and fills the database)
- supported by tableau (info: https://help.tableau.com/current/online/en-us/to_connect_live_sql.htm)
- accessible via internet
- open source

**Question**
Which Database should be used? 
- postgreSQL (compatible with tableau: yes, compatible with node red: ?)

kruegeralex, 2021-07-22:
In principle, nodered is compatible with a large number of SQL databases.

MySQL: https://flows.nodered.org/node/node-red-node-mysql
Postgres: https://flows.nodered.org/node/node-red-contrib-re-postgres

The MySQL node, however, seems to be more actively maintained and has a larger community, which is why I would recommend it.

larswitte, 2021-07-22:
Thank you, then lets go with MySQL

larswitte, 2021-09-23:
In the end we are not implementing this in the near future. I think it still would be a good idea to implement this long term (or implement a tableau web data connector). 
We probably should use a timescaleDB instead of MySQL to minimize complexity.

closed 2023-03-31 by Scarjit