united-manufacturing-hub/united-manufacturing-hub #360: Add database changes for product shadow

JeremyTheocharis, 2021-07-22:
Implement the database changes in https://github.com/united-manufacturing-hub/united-manufacturing-hub/issues/297#issuecomment-880796956

larswitte, 2021-07-26:
**Latest Update of the timescaleDB changes for digital shadow:**
[image]

**Notes:**
- `step_id` instead of `station_id` allows us more flexibility in the naming convention. There are cases, where a differentiation into stations would not make sense but the differentiation into steps would. This allows us to model multistep processes better.
- Quality_class should be deletet: instead the information will be stored in the `productTagTable` or `productTagStringTable`
- The structure of the inheritance table models every single path in the inheritance of parts. While it is easy to understand and use, it might make problems because of possibly to many entries. To make shure it works in the future we should test the use of this table with a few million test data points. To increase perfomance we should make both the `child_uid` and the `parent_uid` to an index.
- Maybe we could use the name `uniqueProductExternalID` instead of `uniqueProductAlternativeID`. 

Thank you @Gornoka, @JeremyTheocharis and @Scarjit for your help!!!

Gornoka, 2021-07-26:
Additional thoughts,

Unique product table as in other schema:
Hypertable time = Id additional partitioning column assetId

Possible double table architecture for inheritance with child of asset and parent of asset table, each as hypertables for faster search performance and quick resolution of immediate relatives, ie in case of a defect I've product quick steps to isolate related products. Whether this is useful and feasible to implement should be evaluated in a benchmark.

For the product/unique product table defaults can be calculated by dividing machine memory through a constant factor (TBD, depending on final table row size) the other tables would be highly dependent on the complexity of the product tree with many child/parent connections yielding the low values.  Maybe it is useful to add an expected complexity field to the asset table, which could scale the time axis for the inheritance tables. To spread these high edge components into more different subtables. This would require a custom time partitioning function and an immutable Value for every asset, since the information in the hypertable on all products manufactured in the asset would be lost on scaling value change.

Gornoka, 2021-08-03:
results from @Scarjit s benchmarks on this topic last week:

1. fastest speed would be achieved with a database specially designed for graphs (no suprise here)
2. implementation with 2 hypertables is more than 10X faster than the current implementation with 1 double indexed table.
for now the performance should be be sufficient with the double hypertable approach, and there would be the benefit of not using multiple databases in tandem. The size of the hypertable may need to be experimented on different example production chains. maybe a test tool could be added to the next milestone. @JeremyTheocharis

closed 2021-09-23 by larswitte