united-manufacturing-hub/united-manufacturing-hub #367: Database model Refactoring for Digital Shadow

Gornoka, 2021-07-23:
Hello Everyone, thanks for the productive meeting this afternoon @Scarjit @JeremyTheocharis @larswitte ,
Below I will reference the branch with the different model design.

[umh Stack-V-dev-0.pdf](https://github.com/united-manufacturing-hub/united-manufacturing-hub/files/6871078/umh.Stack-V-dev-0.pdf)

[image]

Gornoka, 2021-07-23:
The current branch where I have included the ddl etc can be found below.
https://github.com/Gornoka/united-manufacturing-hub/tree/367-DataModelChanges
please bear in mind that this is only WIP and that the DDL does not include the notes included in the images and PDF, especially the Timescale changes. Also the notes are not necessarily adapted to all components.

Gornoka, 2021-07-27:
updated files from yesterday evening

larswitte, 2021-08-03:
The `AssetID` will be used to identify single steps in the digital shadow use case. We should change the name to something more fitting e.g. `UnitID` or `StepID` to allow more flexible use. 
Doing it this way is beneficial, because we can describe one step/location with only one ID. We can still group multiple step ID's together and build bigger units.

Scarjit, 2021-11-19:
@Gornoka The referenced branch: https://github.com/Gornoka/united-manufacturing-hub/tree/367-DataModelChanges is no longer valid, can you give me the DDL file elsewhere ?

Gornoka, 2021-11-19:
Yes I will do so,

closed 2023-03-31 by Scarjit