united-manufacturing-hub/united-manufacturing-hub #375: VerneMQ not starting when using minikube

JeremyTheocharis, 2021-07-26:
Provide a general summary of the issue in the Title above

## Expected Behavior

VerneMQ should start independent of the Kubernetes provider.

## Current Behavior

When using minikube in some instances it fails to start

```log
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
0 0 0 0 0 0 0 0 --:--:-- --:--:-- --:--:-- 0 20763 0 20763 0 0 965k 0 --:--:-- --:--:-- --:--:-- 965k
/usr/sbin/start_vernemq: line 54: warning: command substitution: ignored null byte in input
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
0 0 0 0 0 0 0 0 --:--:-- --:--:-- --:--:-- 0 20763 0 20763 0 0 1192k 0 --:--:-- --:--:-- --:--:-- 1192k
/vernemq/bin/vernemq: line 8: /lib/env.sh: No such file or directory
/vernemq/bin/vernemq: line 140: check_user_internal: command not found
/vernemq/bin/vernemq: line 144: create_pid_dir: command not found
/vernemq/bin/vernemq: line 147: echoerr: command not found
```

## Possible Solution

no idea so far

## Context (Environment)

tested with `main`, `staging` using minikube 

```
minikube version: v1.22.0
commit: a03fbcf166e6f74ef224d4a63be4277d017bb62e
```

JeremyTheocharis, 2021-07-28:
The reason for this was the version jump. Using fixed versions for now.

kruegeralex, 2021-08-03:
VernqMQ still not starting with Tag "latest"

Add Hotfix with fixed Version Number 1.11.0

closed 2021-08-03 by kruegeralex