united-manufacturing-hub/united-manufacturing-hub #390: Error when starting camera connect

JonaVo, 2021-07-28:
Refers branch to 364-add-cameraconnect-to-helm

I started installed the factory-cube edge via helm install and received the following errors in cameraconnect
Please note that I used the branch 364-add-cameraconnect-to-helm on my edge device

[image]

JeremyTheocharis, 2021-07-28:
There is a command in the tutorial, which helps you with uploading the cameraconnect gentilproducer

https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/e4af2b7bc1af0e550f06473d89383c8a408e7eeb/docs/content/en/docs/Developers/factorycube-edge/cameraconnect/index.md#L19

JonaVo, 2021-07-30:
configmap does not work, as we found out together yesterday
Please adjust docs

closed 2021-07-30 by JeremyTheocharis