united-manufacturing-hub/united-manufacturing-hub #395: few minor changes in docs

JonaVo, 2021-07-29:
**1. Please add something like: change namespace in Lens to observe the cluster starting up, otherwise nothing will be displayed in Lens**

> [image]

______________________________________________________________
**2. Add a description how one can get the developement_values.yaml file via wget to simplify the process**

> [image]

______________________________________________________________
**3. Please change to "an edge device"**

> [image]

JonaVo, 2021-07-29:
**Please remove cognex, as only GenICams are supported**

> [image]

Scarjit, 2021-07-29:
Tracked on https://github.com/united-manufacturing-hub/united-manufacturing-hub/tree/395-few-minor-changes-in-docs

PR #499 (2021-09-08): 454 improve quick start
FIxes #454
Fixes #395 
Fixes #472
commit 43851026 (2021-09-08): 454 improve quick start (#499)

FIxes #454
Fixes #395
Fixes #472

* First version

* Updated

* Added development.yaml

* Fixed development.yaml

* Added pictures

* Fixed installation script

* Increased image size

* Improved guide

* Fixed development.yaml

* Improved guide

* This is a good example of a very bad commit message

* Updated docs

* Updated docs

* Updated docs

* Updated docs

* Fixed SSH in SSH guide

* Update

* Finished adding the workshop

* Splitted up Understanding the technologies

* Adjusted search

* Removed search and fixed spelling
closed 2021-09-08 by JeremyTheocharis