united-manufacturing-hub/united-manufacturing-hub #449: `addProduct` messages are not always handled in time

larswitte, 2021-08-16:
**Problem**
Sending `addProduct` MQTT Messges to MQTT-to-postgres should directly create new entries in the `productTable`. However, in my experience, I have to send the `addProduct` messages multiple times, before the entries are created even though entries in the assetTable are created directly.

This is the used NodeRed flow:
[image]
And this is one of the functions:
[image]

Also, after the entries are created, the `id` in the `assetTable` is not SERIAL how it should be.
Data grip:
[image]

Scarjit, 2021-09-02:
"Also, after the entries are created, the id in the assetTable is not SERIAL how it should be."
I can't reproduce this, according to my Datagrip, the DDL shows it being initialized as a serial:
```
-- auto-generated definition
create table assettable
(
    id       serial
        constraint assettable_pkey
            primary key,
    assetid  varchar(20) not null,
    location varchar(20) not null,
    customer varchar(20) not null,
    constraint assettable_assetid_location_customer_key
        unique (assetid, location, customer)
);

alter table assettable
    owner to postgres;

grant select, update, usage on sequence assettable_id_seq to factoryinsight;

grant delete, insert, references, select, trigger, truncate, update on assettable to factoryinsight;
```
Keep in mind, that SERIAL is not an actual Postgres type, but an instruction to tell Postgres, to create an integer type, which is not null and is auto incrementing

Scarjit, 2021-09-02:
MQTT-to-Postgresql seems slow at startup.
After processing a few items, it instantly processes incoming items.

JeremyTheocharis, 2021-09-02:
Apparently, maintenanceactivities is not created.

Here the schema from the current production server:
```sql
-- Table: public.maintenanceactivities

-- DROP TABLE public.maintenanceactivities;

CREATE TABLE IF NOT EXISTS public.maintenanceactivities
(
    id integer NOT NULL DEFAULT nextval('maintenanceactivities_id_seq'::regclass),
    component_id integer NOT NULL DEFAULT nextval('maintenanceactivities_component_id_seq'::regclass),
    activitytype integer,
    "timestamp" timestamp with time zone NOT NULL,
    CONSTRAINT maintenanceactivities_pkey PRIMARY KEY (id),
    CONSTRAINT maintenanceactivities_component_id_activitytype_timestamp_key UNIQUE (component_id, activitytype, "timestamp"),
    CONSTRAINT maintenanceactivities_component_id_fkey FOREIGN KEY (component_id)
        REFERENCES public.componenttable (id) MATCH SIMPLE
        ON UPDATE CASCADE
        ON DELETE CASCADE
        NOT VALID
)

TABLESPACE pg_default;

ALTER TABLE public.maintenanceactivities
    OWNER to postgres;

GRANT ALL ON TABLE public.maintenanceactivities TO postgres;

GRANT ALL ON TABLE public.maintenanceactivities TO factoryinsight;
```

JeremyTheocharis, 2021-09-02:
Found a second table that is not auto created:

```sql
-- Table: public.timebasedmaintenance

-- DROP TABLE public.timebasedmaintenance;

CREATE TABLE IF NOT EXISTS public.timebasedmaintenance
(
    id integer NOT NULL DEFAULT nextval('timebasedmaintenance_id_seq'::regclass),
    component_id integer NOT NULL DEFAULT nextval('timebasedmaintenance_component_id_seq'::regclass),
    activitytype integer,
    intervallinhours integer,
    CONSTRAINT timebasedmaintenance_pkey PRIMARY KEY (id),
    CONSTRAINT timebasedmaintenance_component_id_activitytype_key UNIQUE (component_id, activitytype),
    CONSTRAINT timebasedmaintenance_component_id_fkey FOREIGN KEY (component_id)
        REFERENCES public.componenttable (id) MATCH SIMPLE
        ON UPDATE CASCADE
        ON DELETE CASCADE
        NOT VALID
)

TABLESPACE pg_default;

ALTER TABLE public.timebasedmaintenance
    OWNER to postgres;

GRANT ALL ON TABLE public.timebasedmaintenance TO postgres;

GRANT ALL ON TABLE public.timebasedmaintenance TO factoryinsight;
```

Scarjit, 2021-09-02:
# Conclusion
MQTT-to-Postgres won't work, until the same message is send at least twice.

# Resolution
By integrating tests, we can bypass this behavior.

Scarjit, 2021-09-06:
MQTT-to-Postgres slow startup *seems* to be fixed in #506

closed 2021-09-07 by Scarjit