united-manufacturing-hub/united-manufacturing-hub #455: Improve guide Connecting machines and creating dashboards

JeremyTheocharis, 2021-08-19:
https://docs.umh.app/docs/getting-started/connecting-machines-creating-dashboards/

**Is your feature request related to a problem? Please describe.**
The current guide is not good and very unintuitive.

**Describe the solution you'd like**
We should rewrite it (based on other Node-RED templates, e.g., for brewing)

@kruegeralex Can you provide your current status?

oarne97, 2021-11-22:
After implementing and using the documentation in Venice with Alex's help, I understood what the instructions said. For me personally, the following things in the documentation would have made it more understandable in the first place:

1) At the beginning, simply explain nodes that are normally used: What they are used for? How they are used (screenshots incl. the included data)? What do they do?

2) Example video showing the creation of a Node-RED flow based on an example sensor and showing step by step how to implement it --> time by time we can add more implementation videos this way

3) limit in- and output to mqtt and not use preset in- and outputs --> confused me in the manual
Besides, the flow "small step by step" just makes more sense to me, because otherwise important intermediate steps, which are important for understanding, disappear in a preset node

kruegeralex, 2021-11-22:
@oarne97, thank you for your input. In addition, the sample images of the Node-RED flow should be exchanged, as well as the underlying Node-RED flow that is pre-installed when the stack is installed. Furthermore, a video could be made on how to create a dashboard in Grafana, which could be especially helpful for the Venetian colleagues.

kruegeralex, 2021-11-22:
As an example, you could take a machine from the DCC Aachen, for example the cutting machine, and also add a few photos from the physical implementation.

kruegeralex, 2021-11-23:
@JeremyTheocharis any input?

kruegeralex, 2021-11-23:
**General questions:**
- Who is the target group for the guidance?
- Should it be several small videos or one long one?
- Do we go into detail about Sensorconnect?

**Suggestions:**
- We need a few videos in advance with basic knowledge (e.g. Node-RED, Sensorconnect, MQTT, Grafana etc., JSON parsing).
- Split into several small videos that are didactically arranged so that you can either watch them all one after the other if you have no prior knowledge or watch the one video that helps with a specific problem.
- Videos should be self-contained but tell a consistent story with the videos from before.

kruegeralex, 2021-11-23:
@larswitte What do you think of that? You want to have the videos for a while

oarne97, 2021-11-23:
Also: At the beginning link 2 to 3 general and good node-RED videos, which provide a basis for the tutorial.

oarne97, 2021-11-23:
### first suggestion:

**1. Basic Information** (maybe different spelling)
- refer to 2 to 3 good explanatory videos from note-RED in general, which provide a basic knowledge
- general configuration
- mqtt explanation
- nodes used by us briefly explain individually

**Implementation example** (with short videos if necessary) 
- starting a flow
      - what has to be done
      - what is our starting position
- first node, second node etc. --> building a flow step by step, e. g. at the cutting machine
- cut where it continues with either sensors or machine data --> referring to machine data documentation
- continue with case of sensors

**Create dashboards using factorycube-server with Grafana**
- same procedure
     - explaining the configuration, different types of panels etc.

JeremyTheocharis, 2021-11-23:
> ### first suggestion:
> **1. Basic Information** (maybe different spelling)
> 
> * refer to 2 to 3 good explanatory videos from note-RED in general, which provide a basic knowledge
> * general configuration
> * mqtt explanation
> * nodes used by us briefly explain individually

Looks good! We should also reference to https://docs.umh.app/docs/concepts/node-red-in-industrial-iot/ to motivate people, that they are doing it not with any weird tool, but with something very common. Also to give background information on the nodes that we are using.

We do not need to explain MQTT again. However, we should explain the general dataflow and how we interact with the system. Something like this (unfortunately, I am not able to find the drawio for that):
[image]

I always explain it together with the data model: all raw data is first send to the MQTT broker (without exceptions). We then take the data into Node-RED (or any other microservice), process it, and then write it back again.

> **Implementation example** (with short videos if necessary)
>
> * starting a flow
>   - what has to be done
>   - what is our starting position
> * first node, second node etc. --> building a flow step by [...]

Let us also remove here the default flow as I agree that it is more confusing. I like the idea to do that on one actual physical machine, that makes it more exciting! On a later stage we could also publish it in form of a blog: "How to connect a machine using retrofitting and Node-RED (with photos)"

For each video we should have a explaination in text first (some people just love texts like me, and some love videos like @larswitte)

> **Create dashboards using factorycube-server with Grafana**
> 
> * same procedure
>   
>   * explaining the configuration, different types of panels etc.

larswitte, 2021-11-24:
**Regarding the video tutorials:**
I really like the idea of multiple shorter videos, explaining different techniques in detail.

The most important video in my opinion is one video (as short as possible!) showing the process **from zero to a very basic grafana dashboard**. My first proposal would be something like this: 
- **Low entry bar:** Only one sensor (lets choose one with fast and easy implementation), only one ipc (if possible not the full factorycube hardware), Windows on the Laptop to set everything up and a node red flow as short and easy to understand as possible. The entry bar should be low.
- I would **focus on how** and less on why and what happens in the background. I think it should definetly be possible to get under 20 min maybe under 15 min video length. No lengthy explanation, who we are, what we want to do, what differenciates the UMH stack from other solutions etc., just one very short intro section in the beginning. 
- **Main target group** of this video would be someone who never used (possibly never heared about) the UMH stack, and is interested in it.
- **Main function** of this video could be demonstrating, that you only have to invest something like 15 min or less of research, before you are able to set up a simple system - even with zero prior knowledge. This can boost confidence and lead to more people trying it out. Additionally the video will give the user a fast overview over the essential technologies and how to use them.
- Let's use the **youtube chapter function**, this will allow the user to skip the intro or certain other parts of the video, and directly get to the relevant bit of information.
- **Linking** further videos and articles from us and from third partys in the description.
- **One overview chart:** Physical sensor &rarr; Node-Red (*factorycube edge*) &rarr; database/grafana (*factorycube server*)
- **Title:** Maybe something like: "Set up IoT Network with Node-Red and Grafana based on Kybernetes in under 15 minutes!" and UMH in the thumbnail, description and Channel Name.

kruegeralex, 2021-11-24:
Thank you @oarne97 for your headline ideas I think these can be combined relatively well with @larswitte  video.

The video by @larswitte could be put under the last point of "Basic Information" and thus create a clean transition. I think this video is probably enough for now and the smaller, more detailed videos can be added later and we can focus on the instructions in text form for these more specific topics.

kruegeralex, 2021-11-24:
In Issue #703 , the point was also raised that the IP range of the sensorconnect service must be adjusted so that sensor values are read correctly. I think we should definitely include this in the instructions.

oarne97, 2021-11-26:
### What we have so far:

**1. "Introduction to Node-RED and Grafana"**

_Idea_
- small text what Node-RED is and why Node-RED is used with reference to https://docs.umh.app/docs/concepts/node-red-in-industrial-iot/
- small text what Grafana is and why Grafana is used

→ deeper explanatory documentation for Node-RED and Grafana should not be mentioned at the beginning. A deeper insight will follow later!

**2. "Set up IoT Network with Node-RED and Grafana running on Kybernetes in under 15 minutes!"**

_Idea_
- from zero to a very basic grafana dashboard
- only one sensor → fast and easy implementation
- only one ipc → not a full factorycube hardware
- focus on how and less on why and what happens in the background
- target group of this video is someone who never used (possibly never heard about) the UMH stack, and is interested in it or has its own use case

_How the video will look like_
- setup → Node-RED flow → Grafana
- maybe using a chapter function to skip the intro or to skip to certain other parts of the video

JeremyTheocharis, 2021-11-29:
Everything related to the setup process should go into part 1: https://docs.umh.app/docs/getting-started/setup-development/

Introduction to Node-RED and Grafana fits perfectly in the current chapter that you are working on: https://docs.umh.app/docs/getting-started/connecting-machines-creating-dashboards/

kruegeralex, 2021-11-29:
@JeremyTheocharis Ok, I have then moved the issue to Issue #717

kruegeralex, 2021-11-29:
thx @oarne97!

I would suggest that we prioritise the video as the first section of the guide as this seems to be the most important. After that, the written guide can be developed further.

I would suggest that in the next step you develop the script for the video.

I would suggest that we put 2 data points into Grafana:

Messages under the /count topic 
and
Messages under the /processValue topic

Then you have a contextualised size (count) and a raw size (processValue) so you save the complexity of the machine states and shifts.

kruegeralex, 2021-11-29:
But the datapoints like /state and /shift should definitely be explained in the written instructions with screenshots in the second step.

JeremyTheocharis, 2021-11-29:
I created a YouTube channel and I can provide you access assuming that you have a google account with your company adress

kruegeralex, 2021-12-01:
Summary Interview with Arne:

- Focus of Chapter 2 should be on the video (max 15 minutes from sensor to dashboard).
- Chapter 2 is the supposedly easiest chapter and it therefore makes less didactic sense to make a simulating video here if chapters 0 and 1 have significantly higher entry barriers than chapter 2 -> i.e. people who need the video would not get to chapter 2 at all.

New target state 3 videos:

Chapter 0: Introducing UMH with IT/OT basics (+ target state (Grafana Dashboard, Node-RED, Kubernetes).
Chapter 1: Blunt filming of the installation process
Chapter 2: (As planned) From Sensor to Dashboard

+ More in-depth information then in text form within the individual chapters.

kruegeralex, 2021-12-01:
Script for video in chapter 2:

Setting: White bar table
Camera: Sony Alpha 6500 
Background: Shopfloor and rollup

0 Step: What do you have in the end?
- A sensor that plays data up to a dashboard with videos of the sensor and dashboard.

1 Step: Physical connection and wiring
- Introduction of the components (sensor, io-link master, edge-pc, router, switch, power supply) Using the Factorycube as an example with deepdive to the components.
- Setting up the power supply
- Connecting sensor and io-link master
- Network connection io-link master via switch edge-pc and router
- Connecting laptop to edge-pc via switch and router

2 Step: Software components:
- Download MQTT Explorer
- Start MQTT-Explorer and display sensor data (refer to IP-address where to get it)
- Start Node-RED (refer to IP address where to get it)
- Start Grafana

3 Step:
- Create the flow (TBD: Here go into detail what should be shown and how. IDEE: /count and /processValue)

4: Step:
- Creation of the Grafana dashboard (Alex ask for standard components)

5.
- Show where to download these examples (node-red flow and grafana json)

Idea: After completing the installation, the video plays backwards and comes out at the dashboard created in step 0.

kruegeralex, 2021-12-01:
@JeremyTheocharis any input/ideas?

JeremyTheocharis, 2021-12-01:
Add step to change ip_range if you are using sensorconnect. This will require Lens and therefore some effort in setting up (installing helm on your local PC, etc.). Alternatively, this could be done via SSH with a `helm upgrade factorycube-server --set sensorconnect.ip_range="192.168.10.1/24"` (not tested, just wrote out of my memory). I think alternative 2 is better.

Also I suggest to keep the naming of these chapters consistent as it makes things hard to follow if the name changes constantly. 

Chapter 0: Introducing UMH with IT/OT basics (+ target state (Grafana Dashboard, Node-RED, Kubernetes).
--> 0. Understanding the technologies
Chapter 1: Blunt filming of the installation process --> 1. Installation
Chapter 2: (As planned) From Sensor to Dashboard  --> 2. Connecting machines and creating dashboards

oarne97, 2021-12-01:
@JeremyTheocharis According to you, doesn't the topic of SensorConnect and SHH etc. belong in Chapter 1 (Installation)?

@JeremyTheocharis The naming of these chapters won't change. The video headlines mentioned by alex are only to briefly show the idea of what the video deals with.

JeremyTheocharis, 2021-12-01:
Sensorconnect does not belong in my opinion to installation, it belongs to configuration and usage of the system.

oarne97, 2021-12-01:
**Building 2. connecting machines and creating dashboards**

0. video with above described content

1. introduction
- UMH Hub has a wide range of ports and connectors. (Image graphic with logos of the different standards: OPC/UA, Siemens S7,TCP/IP, Rest API, Modbus, MQTT).
- These are e.g. OPC/UA, Siemens S7,TCP/IP, Rest API, Modbus, MQTT
- These can simply be read out using Node-RED or, in the case of IO-Link, are read out automatically and made available via MQTT.
- With the help of these data points and our data model, the data can be transferred into a standardized data model, independent of the machine type and manufacturer. and make it usable for various data services, e.g. Grafana.

2. node-RED
- Introduce why Node-RED is the standard (reference to article).
- Explain Node-RED on the basis of exemplary cases
1st example: Retrofit with the help of external sensor technology (Sensorconnect incl. adaptation of IP range, MQTT) 
2nd example: connection to interface (OPC/UA).
- General configuration for the examples
- Outputs gone, only MQTT output

3. create dashboards using factorycube-server

--> From now on the content will only be processed in the pull request

oarne97, 2021-12-10:
**Updates:**
- Send count messages only via light barrier
- Machine states are transmitted via the button bar. Button 1: running, Button 2: not running, etc.

Outlook: With complicated logic, the machine state can also be transmitted with light barrier.For reasons of complexity, this will not be explored in depth in our example.

- Transfer inductive sensor as process value/process
- Insert photos
- Make example process downloadable

commit 51b7ca03 (2022-01-20): feat: Improve guide connecting machines and creating dashboards (#678)

Fixes #455
closed 2022-01-20 by JeremyTheocharis