united-manufacturing-hub/united-manufacturing-hub #469: Querying images by Associated product tags and limiting the output number

Gornoka, 2021-08-25:
**Is your feature request related to a problem? Please describe.**
Requesting image data for particular product classes or unknown timeframes results in large overhead in network transfer and parsing time.

**Describe the solution you'd like**
Function to allow querying image metadata with more params such as Number of images received and associated product tags 
( e.g. usefull for querying just images of Scrap or other quality classes)

**Describe alternatives you've considered**
Querying the api and metadata of the current paths leads to much overhead, especailly for applications where only the last image of a given class is interesting, since the time frame to query can not be determined without more informations ( e.g. searching for an image of a rare class, would result in an large amounts of unnecessary metadata)

Scarjit, 2022-06-03:
It this still relevant ?

Gornoka, 2022-06-03:
The situation still persists, so searching images is basically not possible with the current API(AFAIK).

Scarjit, 2023-03-31:
Blobstorage is deprecated

closed 2023-03-31 by Scarjit