united-manufacturing-hub/united-manufacturing-hub #496: Factorysimulator for testdata and integration tests

JeremyTheocharis, 2021-09-02:
**Is your feature request related to a problem? Please describe.**
As a mechanical engineer I want to have example data in the dashboard when starting up the stack for presentations and testing. Furthermore, as a mechanical engineer I would be great to have a reference architecture with sample code

Additionally, as developer I want to ensure that once a functionality is implemented that it does not break in the future (--> Integration test on each PR)

**Describe the solution you'd like**
A microservice called `factorysimulator`, which simulates a nonexisting factory. It needs to use all MQTT messages and cover all edge cases.

This could be either Python microservice or Node-RED in a second instance (prefered).

**Additional context**

Fixes #491 
Fixes #449 
Fixes #432 

Refers to #493

larswitte, 2021-09-02:
This is my current simulation for two production steps in node red to test digital shadow functionality (#297):

[{"id":"e259d00c.c234f8","type":"tab","label":"IntegrationTest1: AddProducts","disabled":false,"info":"Execute this flow first. Maybe set Mqtt out node username to TESTING"},{"id":"23ae98e1.65e538","type":"tab","label":"IntegrationTest2: ExampleStation","disabled":false,"info":"Execute second."},{"id":"d81eab1a.cff018","type":"mqtt-broker","name":"","broker":"factorycube-server-vernemq","port":"1883","clientid":"","usetls":false,"compatmode":false,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""},{"id":"26181dda.eabe72","type":"mqtt in","z":"e259d00c.c234f8","name":"","topic":"ia/factoryinsight/#","qos":"2","datatype":"auto","broker":"d81eab1a.cff018","x":380,"y":80,"wires":[["6f33d2f4.beb6fc"]]},{"id":"6f33d2f4.beb6fc","type":"debug","z":"e259d00c.c234f8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":610,"y":80,"wires":[]},{"id":"3fdff556.e4b17a","type":"inject","z":"e259d00c.c234f8","name":"","props":[{"p":"payload.begin_timestamp_ms","v":"","vt":"date"},{"p":"payload.aid","v":"","vt":"num"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":150,"y":140,"wires":[["5966dff5.e1f848","edbe929d.918a3"]]},{"id":"5966dff5.e1f848","type":"function","z":"e259d00c.c234f8","name":"addProduct ValveHead","func":"msg.payload = \n    {\n        \"product_id\": \"VH_Type123\",\n        \"time_per_unit_in_seconds\": 0.3\n    }\n\n\nreturn msg;","outputs":1,"noerr":0,"initialize":"// to store a variable/ Code added here will be run once\n// whenever the node is started.\n","finalize":"","x":340,"y":140,"wires":[["be44677d.e204b8","79ff2b30.1eb9c4","1afbb195.172d76","6ed30fe6.d5c9d8"]]},{"id":"be44677d.e204b8","type":"mqtt out","z":"e259d00c.c234f8","name":"","topic":"ia/factoryinsight/TestLocation123/storage/addProduct","qos":"","retain":"","broker":"d81eab1a.cff018","x":820,"y":140,"wires":[]},{"id":"6d5ca5c1.9f1a44","type":"comment","z":"e259d00c.c234f8","name":"Info: Add the products","info":"Info: Testprocess like Assy1, station 2, without Gasket selection and without VH history (VH comes from Storage.)","x":160,"y":80,"wires":[]},{"id":"edbe929d.918a3","type":"function","z":"e259d00c.c234f8","name":"addProduct Gasket","func":"msg.payload = \n    {\n        \"product_id\": \"Gasket_Type123\",\n        \"time_per_unit_in_seconds\": 1.5\n    }\n\nreturn msg;","outputs":1,"noerr":0,"initialize":"\n","finalize":"","x":350,"y":320,"wires":[["b6718fe8.1fe51","af45eab4.01f748","3562003.b91088","6ed30fe6.d5c9d8"]]},{"id":"79ff2b30.1eb9c4","type":"mqtt out","z":"e259d00c.c234f8","name":"","topic":"ia/factoryinsight/TestLocation123/Assy1/addProduct","qos":"","retain":"","broker":"d81eab1a.cff018","x":810,"y":200,"wires":[]},{"id":"1afbb195.172d76","type":"mqtt out","z":"e259d00c.c234f8","name":"","topic":"ia/factoryinsight/TestLocation123/Assy1Gasket/addProduct","qos":"","retain":"","broker":"d81eab1a.cff018","x":840,"y":260,"wires":[]},{"id":"b6718fe8.1fe51","type":"mqtt out","z":"e259d00c.c234f8","name":"","topic":"ia/factoryinsight/TestLocation123/storage/addProduct","qos":"","retain":"","broker":"d81eab1a.cff018","x":820,"y":320,"wires":[]},{"id":"af45eab4.01f748","type":"mqtt out","z":"e259d00c.c234f8","name":"","topic":"ia/factoryinsight/TestLocation123/Assy1/addProduct","qos":"","retain":"","broker":"d81eab1a.cff018","x":810,"y":380,"wires":[]},{"id":"3562003.b91088","type":"mqtt out","z":"e259d00c.c234f8","name":"","topic":"ia/factoryinsight/TestLocation123/Assy1Gasket/addProduct","qos":"","retain":"","broker":"d81eab1a.cff018","x":840,"y":440,"wires":[]},{"id":"6ed30fe6.d5c9d8","type":"debug","z":"e259d00c.c234f8","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":680,"y":580,"wires":[]},{"id":"19151a34.a343ae","type":"comment","z":"23ae98e1.65e538","name":"Info: Testprocess like Assy1 but without Gasket selection and without VH history (VH comes from Storage.)","info":"Info: Testprocess like Assy1, station 2, without Gasket selection and without VH history (VH comes from Storage.)","x":400,"y":440,"wires":[]},{"id":"86b91b56.b72fa8","type":"inject","z":"23ae98e1.65e538","name":"","props":[{"p":"payload.begin_timestamp_ms","v":"","vt":"date"},{"p":"payload.aid","v":"","vt":"num"}],"repeat":"130","crontab":"","once":false,"onceDelay":0.1,"topic":"","x":130,"y":560,"wires":[["471760d9.1613f","c18519f1.c9b358"]]},{"id":"471760d9.1613f","type":"function","z":"23ae98e1.65e538","name":"New uniqueProduct from Storage","func":"msg.payload = \n    {\n      \"begin_timestamp_ms\": Date.now(),\n      \"end_timestamp_ms\": null,\n      \"productID\": \"VH_Type123\",\n      \"isScrap\": false,\n      \"uniqueProductAlternativeID\": String(flow.get(\"counterVH\"))\n    }\n\nreturn msg;","outputs":1,"noerr":0,"initialize":"flow.set(\"counterVH\",57000458); // to store a variable/ Code added here will be run once\n// whenever the node is started.\n","finalize":"","x":360,"y":560,"wires":[["af7f8dee.974838","8601a0db.86ebf"]]},{"id":"af7f8dee.974838","type":"debug","z":"23ae98e1.65e538","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload", [... cut ...] er\",1000); // to store a variable/ Code added here will be run once\n// whenever the node is started.\n","finalize":"","x":420,"y":1320,"wires":[["5928a7bc.4c059","ace5904a.d89a28"]]},{"id":"e2c85ce7.7b5","type":"delay","z":"23ae98e1.65e538","name":"","pauseType":"delay","timeout":"1","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":120,"y":1320,"wires":[["81caa86.0b676d8","fa34ffcb.3c2cc8"]]}]

Scarjit, 2022-12-21:
We now have multiple simulators integrated into the stack

closed 2022-12-21 by Scarjit