united-manufacturing-hub/united-manufacturing-hub #501: Updata Digital Shadow Documentation + Openapi spec for get uniqueProductsWithTags

larswitte, 2021-09-02:
**Todo**
- Update Digital shadow (#297) documentation on main page to latest implementation
- add openapi spec for uniqueProductsWithTags (factoryinsight api for digital shadow)

larswitte, 2021-09-29:
https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/edc1a414333f079e4e61477138b8d8896537c6a8/docs/content/en/docs/Concepts/DigitalShadow/index.md#L8

Delete this line

larswitte, 2021-09-29:
Change:
https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/edc1a414333f079e4e61477138b8d8896537c6a8/docs/content/en/docs/Concepts/DigitalShadow/index.md#L19-L20

Rough example:
After that `factoryinsight` provides an interface (Rest API) for formatted data providing maximal  
 usability and easy access for Dashboards or BI-Tools. Especially the Grafana Datasouce of UMH can directly be used to display digital shadow data on a Grafana Dashboard.

larswitte, 2021-09-29:
https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/edc1a414333f079e4e61477138b8d8896537c6a8/docs/content/en/docs/Concepts/DigitalShadow/index.md#L349

massage --> message

larswitte, 2021-09-29:
**Include how to store Image Data:**
The digital shadow itself dooesn't store big images (because timescaleDB not ideal for that), it only stores the keys while the blobstorage holds key value pairs.
If possible, you can directly include the image key as an AID in the Digital Shadow, if not, you can store the key as productTag or productTagString. Then you can get this key from the factoryinsight Rest API and get the image with that key from the blob storage.

*The blob storage is not yet implemented!*

larswitte, 2021-09-29:
https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/edc1a414333f079e4e61477138b8d8896537c6a8/docs/content/en/docs/Concepts/DigitalShadow/index.md#L449

Specify with something like:
5. Return all parent AID's under the column name of the corresponding parent productID's. Return the child AID and UID. Return the productTag and productTagString values under the column name of the corresponding valueNames.

Make Open API spec for that.

larswitte, 2021-09-29:
Document, that this is not implemented in the stack. The end user can use this concept, if necessary.

larswitte, 2021-09-30:
Also: Go throug UMH Datamodel documentation and update Mqtt messages implemented for digital shadow

JeremyTheocharis, 2021-11-25:
> https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/edc1a414333f079e4e61477138b8d8896537c6a8/docs/content/en/docs/Concepts/DigitalShadow/index.md#L8
> 
> Delete this line

Done

> Change:
>
> https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/edc1a414333f079e4e61477138b8d8896537c6a8/docs/content/en/docs/Concepts/DigitalShadow/index.md#L19-L20
>
> Rough [...]

Done

> https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/edc1a414333f079e4e61477138b8d8896537c6a8/docs/content/en/docs/Concepts/DigitalShadow/index.md#L349
> 
> massage --> message

Done

> https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/edc1a414333f079e4e61477138b8d8896537c6a8/docs/content/en/docs/Concepts/DigitalShadow/index.md#L449
>
> Specify with something [...]

Done

> Document, that this is not implemented in the stack. The end user can use this concept, if necessary:
> 
> https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/edc1a414333f079e4e61477138b8d8896537c6a8/docs/content/en/docs/Concepts/DigitalShadow/index.md#L451-L462

Done

PR #714 (2021-11-29): docs: update digital shadow documentation
Fixes #501
commit a19ed626 (2021-11-29): docs: updated digital shadow documentation

Partly fixes #501
closed 2021-11-29 by JeremyTheocharis