united-manufacturing-hub/united-manufacturing-hub #536: As a User i want to install the Stack without using lens

kruegeralex, 2021-09-09:
**Is your feature request related to a problem? Please describe.**
At the Moment you Need Lens to verify a sucessful setup and get the Grafana secrets.

**Describe the solution you'd like**
As a User i want to have:
- the grafana secrets either static oder plotted in the CLI
- a Proof of a sucessful Installation without checking lens

Scarjit, 2022-01-27:
It is possible to get the current secrets by following these commands:

```kubectl get pods -n factorycube-server```
->
```
NAME                                                            READY   STATUS    RESTARTS      AGE
svclb-factorycube-server-vernemq-mmnnf                          0/2     Pending   0             13d
svclb-factorycube-server-grafana-mwb9s                          1/1     Running   0             13d
svclb-factorycube-server-9zlwb                                  1/1     Running   0             13d
svclb-factorycube-server-grafanaproxy-service-ns8kt             1/1     Running   0             13d
factorycube-server-grafanaproxy-79b7f5d867-x6zvd                1/1     Running   0             13d
factorycube-server-grafana-7d65684cc5-t524p                     1/1     Running   2 (13d ago)   13d
factorycube-server-vernemq-0                                    1/1     Running   0             13d
factorycube-server-factoryinput-0                               1/1     Running   4 (13d ago)   13d
factorycube-server-redis-node-0                                 3/3     Running   0             13d
factorycube-server-mqtttopostgresql-0                           1/1     Running   3 (12d ago)   12d
factorycube-server-factoryinsight-deployment-5ffc5cfb7b-spl4f   1/1     Running   0             9d
factorycube-server-factoryinsight-deployment-5ffc5cfb7b-fm4gh   1/1     Running   0             9d
factorycube-server-timescaledb-0                                1/1     Running   0             13d
```

```kubectl exec -n factorycube-server -it POD_NAME -- printenv | grep ENV_VARIABLE_NAME```
e.g
```kubectl exec -n factorycube-server -it factorycube-server-grafana-7d65684cc5-t524p -- printenv | grep GF_SECURITY_ADMIN_PASSWORD```
->
```
Defaulted container "grafana" out of: grafana, init-umh-datasource (init)
GF_SECURITY_ADMIN_PASSWORD=abcdefgh-1337-7331-0000-111111111111
```

To get the proof of successful installation, you would have to do ```kubectl logs POD_NAME``` or wait, for a more centralized logging aggregator.

Scarjit, 2022-12-21:
Resolved using above method, needs to get documented

closed 2023-03-31 by Scarjit