united-manufacturing-hub/united-manufacturing-hub #547: factory cube edge and server can not run on the same device reliably since they both occupy the same port for mqtt

Gornoka, 2021-09-15:
when running factory cube edge and server on the same device both mqtt brokers try to use the same port, this yields unpredictable behaviour since one broker may be faster than the other
## Expected Behavior

Tell us what should happen 
both brokers should use different porst so that factorycube edge and server can run on the same device/ network adapter
## Current Behavior

they both bind on the same port and one randomly crashes 
## Possible Solution

change one to use a different port (probably server since edge has more connections to external software)

## Steps to Reproduce
1. install factorycube edge and server using the install guide ( development config)
2. observe unpredictable behaviour

Gornoka, 2021-09-30:
issue is depending on the exact hardware this runs on.
quick fix from @JeremyTheocharis delete factorycube-server-vernemq service

Scarjit, 2022-03-08:
Fixed with 0.8.X

PR #557 (2021-09-28): fix: MQTT port collisions in development setup between factorycube-se…
…rver and factorycube edge

Fixes #547
commit bdd5c768 (2021-09-28): fix: MQTT port collisions in development setup between factorycube-server and factorycube edge (#557)

Fixes #547
commit 5489de37 (2021-09-28): fix: MQTT port collisions in development setup between factorycube-server and factorycube edge (#557)

Fixes #547
commit c3518004 (2021-09-28): fix: MQTT port collision

Fixes #547
(previous commit 5489de3 did not work)
closed 2022-03-08 by Scarjit