united-manufacturing-hub/united-manufacturing-hub #548: Extremely high CPU utilization without data flow

kruegeralex, 2021-09-16:
Currently, an extremely large amount of resources are used without sending a large amount of data.

Version of the helmet chart: 0.6.0 with scarjit/mqtt-to-postgres:staging

Pod MQTT-to-postgres was deleted incl. the volumes but the problem is only slightly reduced

[image]

[image]

JeremyTheocharis, 2021-09-16:
Apparently, the reason for this is that the system is creating multiple times per second a table called `tmp_processvaluetable` (thank you @pascalwhoop ). We got this information out of the logs of timescaledb.

@Scarjit Do you want to take a look?

Scarjit, 2021-09-16:
Ill check it on my local stack this evening

Scarjit, 2021-09-16:
```tmp_processvaluetable``` is only created by mqtt-to-postgres: https://github.com/united-manufacturing-hub/united-manufacturing-hub/search?q=tmp_processvaluetable

The logic was not modified by the mqtt-to-postgres rework, and was supposed to reduce CPU usage.
What is the frequency of messages hitting that endpoint ?

JeremyTheocharis, 2021-09-17:
It is hard to tell as the log does not provide a resolution under one second. However, it is multiple times per second. Here a log excerpt:

```
			
2021-09-17 13:52:03 UTC [13179]: [61449d83.337b-11] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 3: 
				CREATE TEMP TABLE tmp_processvaluetable 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13183]: [61449d83.337f-4] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 4: 
				CREATE TEMP TABLE tmp_processvaluetable64 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13179]: [61449d83.337b-12] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 3: 
				CREATE TEMP TABLE tmp_processvaluetable 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13182]: [61449d83.337e-3] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 5: 
				CREATE TEMP TABLE tmp_processvaluetable64 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13179]: [61449d83.337b-13] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  disconnection: session time: 0:00:00.071 user=factoryinsight database=factoryinsight host=10.42.0.230 port=44806
2021-09-17 13:52:03 UTC [13172]: [61449d82.3374-15] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  disconnection: session time: 0:00:00.171 user=factoryinsight database=factoryinsight host=10.42.0.230 port=44792
2021-09-17 13:52:03 UTC [13183]: [61449d83.337f-5] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 8: 
				CREATE TEMP TABLE tmp_processvaluetable 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13185]: [61449d83.3381-1] [unknown]@[unknown],app=[unknown] [00000] LOG:  connection received: host=10.42.0.230 port=44818
2021-09-17 13:52:03 UTC [13186]: [61449d83.3382-1] [unknown]@[unknown],app=[unknown] [00000] LOG:  connection received: host=10.42.0.230 port=44820
2021-09-17 13:52:03 UTC [13183]: [61449d83.337f-6] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 8: 
				CREATE TEMP TABLE tmp_processvaluetable 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13182]: [61449d83.337e-4] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 5: 
				CREATE TEMP TABLE tmp_processvaluetable64 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13186]: [61449d83.3382-2] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  connection authorized: user=factoryinsight database=factoryinsight SSL enabled (protocol=TLSv1.3, cipher=TLS_AES_256_GCM_SHA384, bits=256, compression=off)
2021-09-17 13:52:03 UTC [13183]: [61449d83.337f-7] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 8: 
				CREATE TEMP TABLE tmp_processvaluetable 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13182]: [61449d83.337e-5] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 5: 
				CREATE TEMP TABLE tmp_processvaluetable64 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13185]: [61449d83.3381-2] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  connection authorized: user=factoryinsight database=factoryinsight SSL enabled (protocol=TLSv1.3, cipher=TLS_AES_256_GCM_SHA384, bits=256, compression=off)
2021-09-17 13:52:03 UTC [13183]: [61449d83.337f-8] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 8: 
				CREATE TEMP TABLE tmp_processvaluetable 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13182]: [61449d83.337e-6] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 5: 
				CREATE TEMP TABLE tmp_processvaluetable64 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13183]: [61449d83.337f-9] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 8: 
				CREATE TEMP TABLE tmp_processvaluetable 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
[... 409 lines cut ...]
				;
			
2021-09-17 13:52:03 UTC [13192]: [61449d83.3388-10] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 8: 
				CREATE TEMP TABLE tmp_processvaluetable64 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13189]: [61449d83.3385-25] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 3: 
				CREATE TEMP TABLE tmp_processvaluetable 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13196]: [61449d83.338c-6] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 5: 
				CREATE TEMP TABLE tmp_processvaluetable64 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13189]: [61449d83.3385-26] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  execute 3: 
				CREATE TEMP TABLE tmp_processvaluetable 
					( LIKE processValueTable INCLUDING DEFAULTS ) ON COMMIT DROP 
				;
			
2021-09-17 13:52:03 UTC [13196]: [61449d83.338c-7] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  disconnection: session time: 0:00:00.058 user=factoryinsight database=factoryinsight host=10.42.0.230 port=44864
2021-09-17 13:52:03 UTC [13189]: [61449d83.3385-27] factoryinsight@factoryinsight,app=[unknown] [00000] LOG:  disconnection: session time: 0:00:00.175 user=factoryinsight database=factoryinsight host=10.42.0.230 port=44826
```

Scarjit, 2021-09-23:
Fixing this issue, without changing the database structure will be difficult.
We could however, reduce the memory footprint, by dynamically pausing the execution of the insertion "thread", to allow accumulation of more data in the queue, before being processed in postgres.

# Current implementation
The current implementation could result in a state, where each item is handled on its own, therefore generating a large amount of overhead at the db side.

# Proposed Solution
The solution would save the data in a database, designed for inserting low amounts of data at high frequencies.
This would require rewriting in some parts of the stack, and would best we done, when we generally work on the DB.

# Proposed Fix
A hotfix for this problem would be to sleep the worker "thread" for N/Queue length time, where N is a reasonable large number.
Ex:
```
loop {
  len = queue.length
  if len == 0:
    sleep(100ms)
    continue
  processQueue()
  sleep(1000ms/len)
}
```

## Problems with proposed fix
This fix has the problem, that data would not nessecarly be inserted as quick as possible

JeremyTheocharis, 2021-10-15:
This is causing a crash in high availability setups as the database stops responding if three microservices do it.

commit 156ca8de (2021-11-02): fix: default flow in node red causing crash of mqtt to postgresql (#606)

Fixes #560
Fixes #597
Fixes #548

* style: refactored misleading variable names

* fix: MQTT messages will now stay inside StoredRawMQTT if redis or postgres are not available.

This change will await redis and postgres on startup, while already listening for mqtt messages.

* fix: Fixed redis check

* feat: each queue handler will now check if the error is recoverable, re-queuing if so.
Recoverable errors, are:
 - sql: database is closed
 - driver: bad connection
 IsRecoverablePostgresErr sleeps internally, if the error [... cut]
closed 2021-11-02 by Scarjit