united-manufacturing-hub/united-manufacturing-hub #569: mqtt-bridge: add option in Helm chart for connection with Azure IoT Hub

JeremyTheocharis, 2021-10-05:
**Is your feature request related to a problem? Please describe.**
Connecting to Azure IoT Hub using e.g., Node-RED, can be unreliable, especially when having a bad internet connection. It would be great to use mqtt-bridge, but it does not allow authentication via Username / Password yet.

**Describe the solution you'd like**
Enable Username / Password authentication in Azure IoT Hub. Furthermore, add an extra option just for Azure IoT Hub that asks for things like "SAS token" and then automatically converts them to MQTT connection parameters.

Scarjit, 2022-12-21:
Resolved, mqtt-bridge allows for passwords and certificates

Scarjit, 2022-12-21:
nvm, the SAS option is not yet resovled

closed 2022-12-21 by JeremyTheocharis