united-manufacturing-hub/united-manufacturing-hub #626: Use Kafka as message broker inside our stack, with an mqtt bridge for outside communication

Scarjit, 2021-11-02:
**Is your feature request related to a problem? Please describe.**
Using Vernemq or emqx leads to various problems inside our stack, due to there unreliability.
Also using Kafka removes the need for us to buffer incoming messages, per microservice and instead lets Kafka do that.

**Describe the solution you'd like**
Internal services should use Kafka as there message queue.
This requires re-writing of at least mqtt-to-postgres & mqtt-to-blob.

JeremyTheocharis, 2021-12-16:
Move together with #625 and use it as next version v0.8.0 (instead of database, #629 )

## v0.8.0

- one Helm chart
- incl. Kafka
- incl. kafka microservices as preview
- fully working
- basically kafka doing nothing

JeremyTheocharis, 2022-01-18:
> Move together with #625 and use it as next version v0.8.0 (instead of database, #629 )
> 
> ## v0.8.0
> * one Helm chart
> * incl. Kafka
> * incl. kafka microservices as preview
> * fully working
> * basically kafka doing nothing

Let's use the existing kafka-to-blob as testing. v0.8.0 is working when we can set it up, it shows no errors and stores images in the blob storage for 2 days in a row without crashing. Even if it crashes no messages are lost.