united-manufacturing-hub/united-manufacturing-hub #627: MQTT messages should be standardized and versioned

Scarjit, 2021-11-02:
**Is your feature request related to a problem? Please describe.**
Our current MQTT message model is not standardized (ex: some messages have different time formats).

**Describe the solution you'd like**
We should implement an v1 of our message format, with an common standard.
Future breaking changes should require increasing the version number.

I would recommend putting the version into the topic, after the specifier.
Example:
```
ia/raw/2020-0102/0000005898845/X01/210-156
```
->
```
ia/raw/v1/2020-0102/0000005898845/X01/210-156
```

This would allow us to more flexible change the model (Changing the definition of ```raw``` would not change the version of other messages)

JeremyTheocharis, 2021-11-02:
see also https://github.com/united-manufacturing-hub/united-manufacturing-hub/milestone/7

JeremyTheocharis, 2021-11-03:
let's substitute ia with the version number. ia is anyway a relict from old times. If we do this we also have to adjust all the ACLs

Scarjit, 2021-11-03:
> let's substitute ia with the version number. ia is anyway a relict from old times. If we do this we also have to adjust all the ACLs

I don't think we should have the version number as top-level.

1) This would prevent us from changing any sub-topic, without creating a new "major" version:
Example:
 - umh/v1/commandA
 - umh/v1/commandB

If we want to do an change on commandA, we can just publish the new version as:
 - umh/v2/commandA

If we instead have it top-level, this would require us to always increase the version everywhere:
Example:
 - v1/commandA
 - v1/commandB

If we want to do an change on commandA, we will have to also specify that commandB in the new version is the same as in the old version, creating unnecessary overhead.

2) Having just a number as top-level will likely confuse people, since it is not clear, which application the version number belongs to.

Therefor i would change it to:

umh/[VERSION]/[Topic]*
ex:
umh/v1/count

closed 2023-03-31 by Scarjit