united-manufacturing-hub/united-manufacturing-hub #658: Allow multi tenant with entirely separate databases

JeremyTheocharis, 2021-11-10:
**Is your feature request related to a problem? Please describe.**
Some customers might require not sharing their database with other customers.

**Describe the solution you'd like**
Easy way to spawn the Helm Chart multiple times, so that each customer has their own namespace and database

**Describe alternatives you've considered**
Alternatively, you could only modify the services directly interacting with the database. As this involves most of the services running you could also clone the entire Helm Chart multiple times. 

**Additional context**
Based on expert discussions.

Scarjit, 2022-06-03:
Is this still planned ?

Scarjit, 2023-03-31:
Is this still relevant @JeremyTheocharis  ?

closed 2024-10-22 by JeremyTheocharis