united-manufacturing-hub/united-manufacturing-hub #660: Long time between MQTT message and readable message in dashboard (>30 seconds)

kruegeralex, 2021-11-11:
For an impressive impression, the time between the arrival of a message and its display in the dashboard should be <<5 sec

Please contact @kruegeralex  to get access to the test environment.

JeremyTheocharis, 2021-11-12:
Could you describe further what is happening? Used versions? CPU load? (Previous release had high CPU usage, probably system was not upgraded yet)

kruegeralex, 2021-11-17:
[image]

Helm chart version 0.6.0 with mqtt-to-postgres upgraded to 0.7.0 @Scarjit correct me if i am wrong

Scarjit, 2021-11-18:
I assume the messages get stuck due to a large amount of invalid messages.
The solution will be to tweak the sleep times, but this will also increase CPU load, due to the increased amount of times it tries & failes to insert incorrect data.

Scarjit, 2021-11-19:
See #671

PR #690 (2021-11-24): feat: fix long time between mqtt message and readable message in dashboard
Attempts to resolve #660 by introducing exponential backoff time.
See also #671 & #689
closed 2022-01-27 by Scarjit