united-manufacturing-hub/united-manufacturing-hub #661: MQTT Bridge fails to build up connection

kruegeralex, 2021-11-11:
Please contact @kruegeralex to get access to the test environment
[factorycube-edge-mqttbridge-0.log](https://github.com/united-manufacturing-hub/united-manufacturing-hub/files/7520665/factorycube-edge-mqttbridge-0.log)
t.

JeremyTheocharis, 2021-11-12:
This issue is likely caused by a LevelDB corruption. Delete the PersistentVolume of mqtt-bridge.

This issue is already known. Will link to the other once I am back from holidays.

kruegeralex, 2021-11-18:
How do i delete the persistent Volume? 
[image]

With the procedure described in the picture, the persistent volume disappears from the list for a short time, but then reappears with an age of 154 days.

JeremyTheocharis, 2021-11-19:
I'll write a guide answering this question

Scarjit, 2022-06-03:
MQTTBridge is no longer part of the active stack

closed 2022-06-03 by Scarjit