united-manufacturing-hub/united-manufacturing-hub #664: Rewriting Sensorconnect: Switch from Python to Golang

larswitte, 2021-11-17:
**Overarching Goal**
The current ``sensorconnect`` service allows connection of IO-Link devices to the united-manufacturing-stack. It was written in Python and does not conform anymore to the high accuracy and availability standards of the project. Now we want to completely rewrite the service in Golang and longterm work on better connectivity to different sensors.

**Issue definition**
This issue will be the central one for the ``sensorconnect`` rewriting. The service might be broken down in different issues. All of those should be linked back to this issue.

larswitte, 2021-11-17:
**Open Challenges**
- Using ``IODD.xml`` files (from device manufacturer, specify IO-Link device data): 
   - automatic downlad (if not possible e.g. bad connection --> manual ``IODD.xml`` file transfer in folder)
   - xml parsing in golang
   - different formats from different manufactuers (research IO-Link standard) - *low priority*
- Stable connection in case of changing sensors/IO-Link masters etc. - *high priority*
   - check for changed devices every 10 s
- High maximal bandwidth and number of sensors - *medium priority*
   - how to build test setup?
   - priorization of different devices?

JeremyTheocharis, 2021-11-17:
Regarding XML parsing:

The scope should focus at first on our common sensors from ifm and our button bar. We should explicitly exclude the "weird stuff" like exporting raw vibration data from the VVB001. On a later stage we can incorporate and test more and more sensors / broaden our test range.

larswitte, 2021-11-17:
> Regarding XML parsing:
> 
> The scope should focus at first on our common sensors from ifm and our button bar. We should explicitly exclude the "weird stuff" like exporting raw vibration data from the VVB001. On a later stage we can incorporate and test more and more sensors / broaden our test range.

Thanks Jeremy, I added it to the comment above.

larswitte, 2021-11-17:
**MQTT Messages from sensorconnect**
The current format of MQTT Messages (going from sensorconnect to the broker) is not ideally documented and looks (if I understand the Python code correctly) like this:

```.js
{
  "serial_number": ...,
  "timestamp_ms": ..., // int
  "type": "IO-Link", // in case of IO-Link
  "connected": 1, // 1 if connected
  "<value1Name>": <convertedValue1>,
  "<value2Name>": <convertedValue2>,
...
  "<valueNName>": <convertedValueN>
}
```

If the device is not known from one of the ``IODD.xml`` files, the received valueString is directly written into the JSON message:
```.js
}
  "serial_number": ...,
  "timestamp_ms": ...,
  "type": "IO-Link", //in case of IO-Link
  "connected": 1, // 1 in case of connected
  "value_string": <unconvertedValue>
}
```

If the connection can not be established:
```.js
{
  "serial_number": ...,
  "timestamp_ms": ...,
  "type": "IO-Link", //in case of IO-Link
  "connected": 0, // 0 in case of not connected, also no value included in JSON
}
```

We are going to drop the "serial_number" entry out of all the messages in the new sensorconnect.
Because of legacy reasons we might stick with the resulting format. If we change the datamodel long term, something like the following might be a good option (static key names, breaking mutliple values from one sensor and one time into multiple messages):
```.js
}
  "timestamp_ms": ...,
  "type": "IO-Link", //in case of IO-Link
  "connected": 1, // 1 in case of connected
  "valueName": <nameOfValue1>,
  "value": <Value1>
}
```

```.js
}
  "timestamp_ms": ...,
  "type": "IO-Link", //in case of IO-Link
  "connected": 1, // 1 in case of connected
  "valueName": <nameOfValue2>,
  "value": <Value2>
}
```

larswitte, 2021-12-01:
As far as I know we are switching from MQTT to Kafka.
Would it be ok to directly switch to Kafka with the new Sensorconnect? (@JeremyTheocharis) 
Otherwise we would have to either implement sensorconnect one time with MQTT and one time with Kafka or implement both at the same time into sensorconnect.

JeremyTheocharis, 2021-12-01:
I think moving from MQTT to Kafka will be quite fast, so let's start with MQTT first.

larswitte, 2021-12-22:
## Overview
This is the overview of the planned new sensorconnect microservice:
[image]

## Open Questions
- Should I use the `encoding/gob` package to save structs persistantly (ChachedIoddFiles)? This, as far as I understand it, would allow me to directly save structs including some extra information gained from e.g. ioddfinder. If not, I can try to save the xml files itself. If I save the xml files, this might also make the ioddfromfile function unnecessary. Additionally it would be easier to access and verify the correct download and storage of iodd files.

JeremyTheocharis, 2021-12-22:
I think storing it as a xml file will be easier:

There can be various methods to import the files (ioddfinder or manual), which saves the ioddfromfile function (ioddfinder saves the xml file, which is then loaded into the cache). The cache itself can then be a struct or whatever you need, but it should remain in RAM (no caching on disk)

larswitte, 2021-12-23:
## Updated Overview
*Version 02*
[image]

PR #805 (2022-02-11): feat: rewritten sensorconnect
# Description

Created PR to track @larswitte changes

Fixes #664

## Type of change

Please delete options that are not relevant.

- [ ] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] This change requires a documentation update

# How Has This Been Tested?

Please describe the tests that you ran to verify your changes. Provide instructions so we can reproduce. Please also list any relevan [... cut]
closed 2022-03-08 by Scarjit