united-manufacturing-hub/united-manufacturing-hub #670: Add MQTT-Kafka bridge

Scarjit, 2021-11-19:
**Describe the solution you'd like**
Since external solutions still rely on MQTT, we need an bridge to convert MQTT Messages <-> Kafka Events.

This bridge could be a Kafka plugin or an seperate microservice

JeremyTheocharis, 2021-11-19:
It might also make sense to communicate between distant kafka brokers (one the factory, one in the cloud somewhere) via MQTT because one does not want to share all data between factory and the cloud. No fixed opinion here, just throwing it into the idea bucket.

Scarjit, 2021-11-19:
> It might also make sense to communicate between distant kafka brokers (one the factory, one in the cloud somewhere) via MQTT because one does not want to share all data between factory and the cloud. No fixed opinion here, just throwing it into the idea bucket.

This should be possible by setting it up like this:

Stack 1 [Internal communication in Kafka] -> [MQTT-Kafka-Bridge] -> Some MQTT Broker (ex: In azure or on premise at stack 1 / 2) -> [MQTT-Kafka-Bridge] -> Stack 2 [Internal communication in Kafka]

Scarjit, 2021-11-19:
We might want to have some kind of config to control which MQTT/Kafka messages will be bridged.

Scarjit, 2021-11-19:
Blocked by #674

PR #676 (2021-11-24): feat: Add mqtt kafka bridge
This pull requests adds an bi-directional mqtt <-> kafka bridge, allowing us to change to kafka internally without loosing outside connection.

This pull request *should* not be merged, until #679 is merged into staging

Fixes  #670
commit bd664dbd (2021-11-24): feat: Add mqtt kafka bridge (#676)

Fixes #670 

* feat: Testing implementation for MQTT<->Kafka bridge

* fix: Added build-base dep, removed GCO since we got an c dep and added musl tag

* feat: better caching

* fix: working kafka connection

* feat: MQTT -> Kafka works :)

* fix: added group id & kafka listen topic

* feat: bi-directional mqtt-kafka bridge with tiered caching for recursion hash check

* feat: Kafka message printer

* fix: fixed caching issue

* feat: Added mqtt-kafka-bridge to values.yaml

* fix: changed repo to umh

* fix: reduced log spam
closed 2021-11-26 by Scarjit