united-manufacturing-hub/united-manufacturing-hub #674: MQTT & Kafka messages should have sender identifiers

Scarjit, 2021-11-19:
While creating a test implantation I have run into the following issue:

An external service sends an MQTT messages to my MQTT broker.
The message is received by MQTT-Kafka-Bridge and send to Kafka.

The MQTT-Kafka-Bridge Kafka subscriber will get the message back from Kafka and send it to the MQTT broker again.
This will trigger the receive event in MQTT-Kafka-Brige and the message will then loop forever.

## Possible solution
Add sender identifier to any MQTT & Kafka message.
I would recommend the following structure:
```json
{
  "send_by": "UUID-4 of original sender",
  "bridged_by": [
    "UUID-4 of first bridge",
    "UUID-4 of 2ndbridge",
    "...."
  ]
}
``` 

This would allow any bridge to check if the message was already processed by it and discard it, while allowing networks with multiple bridges.
Any non bridge consumer can just ignore these keys.

The overhead of this solution quite big for simple messages, since it would effectively double most message sizes.

Since i have no good other solution, i would love to hear, anybody else comes up with an better solution.

JeremyTheocharis, 2021-11-19:
The loop can be avoided by checking whether the received message is equal to one in the last 10 seconds.

Proposal:
When receiving a message from MQTT, hash it using a quick, but not necessary cryptographically secure function. Compare with redis whether the hash exists and if not store that hash in redis with a lifetime of 10 seconds.

Scarjit, 2021-11-22:
I'v added tiered caching for the check (Since we might process a large number of messages, i prefer to have it be checked in memory cache first, to save 1 tcp call)

# GetTiered
https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/a79fd966a8df1b2331f182ea66e8239d32b0e91f/golang/internal/cache.go#L725-L744

# SetTiered
https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/a79fd966a8df1b2331f182ea66e8239d32b0e91f/golang/internal/cache.go#L746-L750

There are also 2 new helper functions ([SetTieredLongTerm](https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/a79fd966a8df1b2331f182ea66e8239d32b0e91f/golang/internal/cache.go#L753), which uses redis default data expiration time, can later be used as drop in for other cache calls && [SetTieredShortTerm](https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/a79fd966a8df1b2331f182ea66e8239d32b0e91f/golang/internal/cache.go#L758) which uses memory default expiration time, used by my recursion check)

closed 2021-11-22 by Scarjit