united-manufacturing-hub/united-manufacturing-hub #703: Data from IO Link gateway not sent

ElPipila, 2021-11-24:
## What Should Happen
According to the documentation, sensor readings should be automatically sent to the broker when connecting sensors via IO Link.

## Current Behaviour
No data in the /ia/raw topic

## Possible Solution
Checking the code of the sensor_connect container?

## Steps to Reproduce
1. Install the system following the steps in the documentation
2. Connect an IO Link gateway to the network
3. Connect a sensor to the gateway

## Context (Environment)
1. Factory Cube with a fresh installation of k3os.
2. The container of the image rancher/klipper-lb:v0.2.0 does not execute due to a port conflict presumably with the vernemq/vernemq:1.11.0 container.

I have made sure that the IO Link gateway is in the network. I have restarted the system more than once and nothing solves the problem.

How has this issue affected you? What are you trying to accomplish?
I am trying to create get sensor data over IO Link and to create a dashboard using Grafana.

JeremyTheocharis, 2021-11-24:
You need to specify the IP range, where sensorconnect should look for gateways first. @larswitte and @oarne97 are currently rewriting the documentation, so that is easier to understand.

You can do that by accessing the stack via Lens and changing under apps --> releases the configuration under sensorconnect.iprange

See also: https://docs.umh.app/docs/developers/factorycube-edge/sensorconnect/#ip_range

You might struggle with Lens (xxx not found), for which you can find a solution here: https://docs.umh.app/docs/tutorials/united-manufacturing-hub/working-with-helm-and-lens/#troubleshooting

btw: `The container of the image rancher/klipper-lb:v0.2.0 does not execute due to a port conflict presumably with the vernemq/vernemq:1.11.0 container.` will be solved by #559

kruegeralex, 2021-11-24:
@JeremyTheocharis can you provide some screenshots on how to change the ip range to issue #455? This could help @oarne97

ElPipila, 2021-11-25:
Closing this issue as measures have been taken.

closed 2021-11-25 by ElPipila