united-manufacturing-hub/united-manufacturing-hub #708: Add a pull request template

kruegeralex, 2021-11-25:
Similar to our template for creating an issue, I propose to add a pull request template to fulfil all the suggestions Github has for the community.

see:
- https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/creating-a-pull-request-template-for-your-repository
- https://github.com/united-manufacturing-hub/united-manufacturing-hub/community

Here is an example:

# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes # (issue)

## Type of change

Please delete options that are not relevant.

- [ ] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] This change requires a documentation update

# How Has This Been Tested?

Please describe the tests that you ran to verify your changes. Provide instructions so we can reproduce. Please also list any relevant details for your test configuration

- [ ] Test A
- [ ] Test B

**Test Configuration**:
* Firmware version:
* Hardware:
* Toolchain:
* SDK:

# Checklist:

- [ ] My code follows the style guidelines of this project
- [ ] I have performed a self-review of my own code
- [ ] I have commented my code, particularly in hard-to-understand areas
- [ ] I have made corresponding changes to the documentation
- [ ] My changes generate no new warnings
- [ ] I have added tests that prove my fix is effective or that my feature works
- [ ] New and existing unit tests pass locally with my changes
- [ ] Any dependent changes have been merged and published in downstream modules

Scarjit, 2021-11-25:
I think this has to be merged into main to work, so for now this should stay open

Scarjit, 2021-11-26:
Merged into main, closing

Scarjit, 2021-11-26:
@JeremyTheocharis The template is not shown, when creating a new pull request.
Do you have to enable it somewhere ?

PR #711 (2021-11-25): feat: Added pull request template
# Description

This pull requests adds an pull request template.

Fixes #708

## Type of change
- [X] New feature (non-breaking change which adds functionality)

# How Has This Been Tested?
I'v rendered the markdown, to check for syntax errors

# Checklist:

- [X] I have performed a self-review of my own code
- [X] My changes generate no new warnings
- [X] I have checked that my issue complies with the [Contributing guidelines](https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/main/CONTRIBUTING.md)
commit 6d933909 (2021-12-14): fix: moved pull request template in accordance with https://stackoverflow.com/a/62613198 (#759)

Fixes #708
closed 2021-12-14 by JeremyTheocharis