united-manufacturing-hub/united-manufacturing-hub #731: Extension of factoryinsight for time series TARGET vs ACTUAL production

kruegeralex, 2021-11-30:
In many production facilities, the currently used and desired diagram is a time series that compares the TARGET with the ACTUAL production.

See figure:
[image]

As a user of the United Manufacturing Hubs, I would also like to be able to use this visualisation.

To do this, I only want to tell the system the current products produced via an MQTT topic ("/count"?) and in some way also the target speed.
It would also be interesting to be able to set up the chart so that the target speed is extrapolated and, if necessary, the sum of the target production for this shift is displayed in the same chart. 

This feature was requested by a partner company.

kruegeralex, 2021-12-08:
@JeremyTheocharis , @Scarjit any idea how much work would be involved to integrate this solution?

Scarjit, 2021-12-08:
That should be doable.
orderTable already has the target_units for a time range.
productTable has time_per_unit_in_seconds.

Scarjit, 2022-01-10:
@kruegeralex Is this what you are looking for:

[image]

This would allow to extract the produced items per order, scrapped items, target items (as defined at order creation), and the maximum producible in this timeframe (product table has time per unit)

Scarjit, 2022-01-10:
Implementing this for a shift instead of a single order would also be easily possible, by getting all orders inside the shift first

kruegeralex, 2022-01-10:
[image]

kruegeralex, 2022-01-10:
[image]

kruegeralex, 2022-01-12:
Here an idea for a test set: 
[image]

kruegeralex, 2022-01-13:
[image]

PR #910 (2022-04-19): feat: Rebased #792 onto latest main branch
# Description

This pr provides an extension to factoryinsight, for retrieving the actual vs target production.
The latest version of this pr is also on [docker hub](https://hub.docker.com/repository/docker/scarjit/factoryinsight)

Fixes #731 

This pr superseds #792 by re-basing it on latest main

## Type of change

- [X] New feature (non-breaking change which adds functionality)

# How Has This Been Tested?

- [X] Checked HTTP response with Insomnia
- [x] Data appears correctly in Grafana

**Test Configuration**:
* Kubernetes Version: 
* Helm Versio [... cut]
closed 2022-04-19 by Scarjit