united-manufacturing-hub/united-manufacturing-hub #743: Add logic: remove message from log once processed from all consumer groups

JeremyTheocharis, 2021-12-01:


Scarjit, 2021-12-13:
According to the Kafka docs, the default retention.ms is 7 days.

If we really want to make sure, that everybody has processed the message, we would need another database to track the processing progress of each microservice.
Alternatively we could just assume that microservices, that take more then 7 days to process the message are not in service anymore.

JeremyTheocharis, 2021-12-13:
Maybe do a rough calculation in which scenarios 7 days will be too much (e.g., sending 5000 messages per second might result in unnecessary large storage capacity)

Scarjit, 2021-12-13:
We can add an storage limit, but that would need documentation. (e.g: about the risk of setting the value to low)

JeremyTheocharis, 2021-12-14:
We need to document it either way. Maybe let's leave it as it is currently

closed 2021-12-14 by JeremyTheocharis