united-manufacturing-hub/united-manufacturing-hub #753: Install fails

PLCMercenary, 2021-12-08:
### Expected Behavior

Fully automated installation on K3os or Dev install with minikube should result in an installed and working configuration 

### Current Behavior

Fully automated installation on K3os AND Dev install with minikube fails  on 4 separate instances 2 hardware and 2 Virtual all fail to install factorycube-edge with [unable to recognize "": no matches for kind "Role" in version "rbac.authorization.k8s.io/v1beta1", unable to recognize "": no matches for kind "RoleBinding" in version "rbac.authorization.k8s.io/v1beta1"]

### Relevant log output

```shell
[unable to recognize "": no matches for kind "Role" in version "rbac.authorization.k8s.io/v1beta1", unable to recognize "": no matches for kind "RoleBinding" in version "rbac.authorization.k8s.io/v1beta1"]
```

### Steps to Reproduce

install k3os to disk with cloud init from https://www.umh.app/development.yml, 

install fails after restarting, connecting to NTP servers and upon trying to install factorycube-edge

### Context (Environment)

fails on 2 separate hardware kits x64 machines. also fails on virtual installation as well as fails the minikube instructions for dev installation

### Version

v0.7.x

PLCMercenary, 2021-12-08:
got around it right after I sent this issue. changed the emqx_rbac template slightly from v1beta1 to v1 and I am up using minikube on a virtual machine. not sure how to get through the cloud init just yet. do you need a  pull request?

Scarjit, 2021-12-08:
@PLCMercenary A pull request would be great.
@JeremyTheocharis This needs a hotfix release.

JeremyTheocharis, 2021-12-08:
I am currently trying to reproduce the issue using `k3OS v0.21.5-k3sr1`, but I struggle (it runs through). Which version of Kubernetes and/or k3OS are you using?

My hypothesis: there is a newer version of Kubernetes out there that deprecated v1beta1 and we have been testing it with an older version so far. 

Thank you a lot for the PR btw!

PLCMercenary, 2021-12-08:
> 

@JeremyTheocharis  k3os is v0.22.2-k3s2r0 and the Kubernetes version is  v1.21.7+k3s1,  initially I was using v1.22.4+k3s1 and that also did not build 

You are welcome.. Its my first pr..  but I am just happy to think that I might be getting decent enough at containers and linux admin that I might contribute !!

JeremyTheocharis, 2021-12-08:
Alright, I can confirm that this error appears using `v0.22.2-k3s2r0` and it does not appear using the latest stable version of k3os `v0.21.5-k3sr1`. I will test in the upcoming days whether your PR resolves this issue, thank you for that!

commit 66c8192d (2021-12-16): fix: install fails in k3os prerelease and newer Kubernetes versions due to deprecated function (#754)

Fixes #753
closed 2021-12-16 by JeremyTheocharis