united-manufacturing-hub/united-manufacturing-hub #758: GitGuardian: API key not configured

Scarjit, 2021-12-13:
### Expected Behavior

GitGuardian runs on commit.

### Current Behavior

GitGuardian fails due to missing API keys

### Relevant log output

```shell
Traceback (most recent call last):
  File "/app/.venv/bin/ggshield", line 33, in <module>
    sys.exit(load_entry_point('ggshield', 'console_scripts', 'ggshield')())
  File "/app/ggshield/cmd.py", line 229, in cli_wrapper
    return_code: int = cli.main(standalone_mode=False)
  File "/app/.venv/lib/python3.9/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/app/.venv/lib/python3.9/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/app/.venv/lib/python3.9/site-packages/click/core.py", line 1656, in invoke
    super().invoke(ctx)
  File "/app/.venv/lib/python3.9/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/app/.venv/lib/python3.9/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/app/.venv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/app/ggshield/cmd.py", line 121, in scan
    ctx.obj["client"] = retrieve_client(ctx)
  File "/app/ggshield/utils.py", line 248, in retrieve_client
    raise click.ClickException("GitGuardian API Key is needed.")
click.exceptions.ClickException: GitGuardian API Key is needed.
```

### Steps to Reproduce

Create a new pull request and wait for the git actions to run.

### Context (Environment)

This prevents me from easily checking if any new pull requests expose secrets.

### Version

not applicable

JeremyTheocharis, 2021-12-13:
This is caused if doing PRs from outside (e.g., from forks). Maybe move this issue to wontfix? Your opinion?

Scarjit, 2021-12-13:
Ill tag it here as wontfix and message GitGuardian about it

Scarjit, 2021-12-14:
Awaiting reply from GitGuardian

Scarjit, 2023-03-31:
Still no reply