united-manufacturing-hub/united-manufacturing-hub #760: "Sign" UMH Datasource in Grafana

kruegeralex, 2021-12-13:
Currently the data source is indicated as unsigned. To remove this badge and get visibility in the Grafana Appstore we should sign the data sources and offer them via the Appstore.

Scarjit, 2021-12-13:
According to the [grafana documentation](https://grafana.com/docs/grafana/latest/developers/plugins/sign-a-plugin/#plugin-signature-levels) this would require us to get a Commercial plugin level. (Our stack is open-source, but for-profit)

JeremyTheocharis, 2021-12-13:
We should still ask them, because you do not need to pay to use it (according to the link it only matters if you need to pay for it and not if it is for profit). 

```
Community

No;
Free of charge
```

@Scarjit Do you want to ask them?

Scarjit, 2021-12-14:
E-Mail to grafana was send, awaiting reply

Scarjit, 2022-01-18:
Send another E-Mail :(