united-manufacturing-hub/united-manufacturing-hub #762: VerneMQ can't find free ports on startup [Pending state]

edHiga, 2021-12-13:
### Expected Behavior

svclb-factorycube-server-vernemq POD starting after some minutes with no errors 

### Current Behavior

Using LENS:

svclb-factorycube-server-vernemq keeps on "pending" state, with event status 

"0/1 nodes are available: 1 node(s) didn't have free ports for the requested pod ports."
[image]

### Steps to Reproduce

As we're using a local DB seems that during 2021-12-11 we had 2 issues:

- Disk space (factorycube-server-vernemq) event log was returning Disk Pressure error and 

- Port not available (svclb-factorycube-server-vernemq)

We removed some logs to reduce disk usage before working on how to cycle data on the postgresDB (maybe dump to another file in the future)
After the log removal first error was gone but we're still facing internal port issues

### Context (Environment)

We're currently running Unted Manufactuting Hub on a food industry to send/receive data using HTTP Post and MQTT to azure in order to display data using Grafana

### Version

not applicable

JeremyTheocharis, 2021-12-14:
The error message `VerneMQ can't find free ports on startup [Pending state]` will happen in development setups, where you have data acquisition and data storage (factorycube-edge and factorycube-server) on the same device (we've already talked about it). The warning only rarely indicates any issues and will be removed in further updates. You can find an explanation why this is not important further down.

Based on the email you send I am assuming that you created this issue not only because this warning is appearing, but because something is not working (maybe the dashboard?). **Can you elaborate on that?**

## Explanation
`factorycube-edge` and `factorycube-server` have both their own MQTT broker. Both MQTT broker are accessible using their internal DNS name and will additionally try to expose the port 8883. As this port can only occupied by once (usually by the one `factorycube-edge`), the other throws a warning saying it cannot use port 8883. Therefore, it is not possible to access `factorycube-server` from outside, but it is still possible for internal microservices to access it via the internal DNS name.

edHiga, 2021-12-14:
> The error message `VerneMQ can't find free ports on startup [Pending state]` will happen in development setups, where you have data acquisition and data storage (factorycube-edge and [...]

Many Thanks Jeremy,
System returned to normal operation 40 minutes after disk cleanup and reboot

Part that was not working:
Data was not persisted on postgres, we were querying data using (pgAdmin) - NodeRed and Grafana were accessible and working as expected (old data was there too)

We'll discuss internally (involving the final team that will maintain this) and have a definite answer on how to move to the original and of course more reliable solution from umh. When we started this solution there was no conectivity avaliable on site (now we can say that we partially have).

closed 2021-12-16 by JeremyTheocharis