united-manufacturing-hub/united-manufacturing-hub #771: Bug Report: first 12 addProduct messages do not work

onurcankaraman, 2021-12-16:
### Expected Behavior

When sending a addProduct MQTT message it should be stored in the database

### Current Behavior

The first 12 messages are not stored, probably because the internal database id starts with 1 and there are already 12 elements in the database

### Steps to Reproduce

Install stack
Send 13 messages of addProduct
Only the last message will be stored

### Context (Environment)

Working with the development setup

### Version

v0.7.x

JeremyTheocharis, 2021-12-17:
@Scarjit let's combine it with your current database rework

Scarjit, 2022-01-14:
Reproduced on a fresh database

PR #801 (2022-01-21): fix: default values for productTable are inserted incorrectly
# Description

Our test data inserted products, while also setting there ID.
Doing this, won't change the postgres sequence, resulting in errors, when inserting new values afterwards.

Fixes #771 

## Type of change

- [X] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

I have created a local table, with an id (serial) and name.

### Test A
Select currect sequence ID -> 1
`SELECT last_value FROM test_table_id_seq;`

Insert 3 times by specifing the id to insert at.
`INSERT INTO test_table (id,name) VA [... cut]
commit 490da8c5 (2022-01-21): fix: default values for productTable are inserted incorrectly (#801)

Fixes #771
closed 2022-01-24 by Scarjit