united-manufacturing-hub/united-manufacturing-hub #853: MQTT Broker does not respond

onurcankaraman, 2022-02-11:
### Expected Behavior

In the updated version of umh, MQTT Broker should stay in "connected" status on the nodered screen. So  the data should be sent to Grafana via nodered. It should be also possible to connect to MQTT Broker via MQTT explorer and i should see the data flow on explorer. 

### Current Behavior

In the updated version of umh, MQTT Broker stays in "connecting" status on the nodered screen. So ı can not send data to Grafana via nodered. MQTT Broker appears in running status on Lens. When I want to connect to MQTT Broker via MQTT explorer nothing appears on screen. 

**_Images_**

[image]
[image]
[image]

### Steps to Reproduce

1. Complate Installation.
2. Connect Lens, Grafana and NodeRed
3. Try Connect MQTT Broker

### Context (Environment)

Installed k3os via default development.yaml on IPC. Installations completed succesful but no data can be send to grafana via nodered. 

### Version

not applicable

JeremyTheocharis, 2022-02-14:
Which username and topic are you using in MQTT broker? it should be username `ia_nodered` and topic `ia/#`

onurcankaraman, 2022-02-15:
It should be whatever is assigned as the default value, I did not make any manual changes. But if you tell me where to check, I can check and get back to you immediately.

By the way, I noticed something while I was playing around, but still no solution. When I'm tampering with the MQTT Output node in Node Red, if I update the protocol to MQTT V3.1(Legacy), I only see the text "Connection Established" on nodered. But when I send data, no data goes to grafana. At the same time, I cannot see the data flow through MQTT Broker.

**IMAGES**
[image]
[image]

Scarjit, 2022-02-15:
Please goto the security tab and use the values provided above by @JeremyTheocharis

JeremyTheocharis, 2022-02-16:
[image]
This security tab :)

onurcankaraman, 2022-02-17:
When I entered the user name, the data flow started. I can see the data in Grafana but not on MQTT Explorer.

[image]

Scarjit, 2022-02-17:
@onurcankaraman Which Username and MQTT Client ID have you set in MQTT Explorer ?

onurcankaraman, 2022-02-17:
I don't know. When I loaded the vernemq certificates from Lens software to mqtt explorer (I saved all of them in a notepad and selected them one by one from the advanced menu), the problem was solved.
[image]

onurcankaraman, 2022-02-24:
I am having the same issue again with MQTT Explorer. Even though I installed the certificates, the data does not appear here. There is data flow between NodeRed and Grafana. It just doesn't show up here.

[image]
[image]
[image]

Scarjit, 2022-02-24:
@onurcankaraman Is there any difference between what you have tried today, and what worked in your last post ?

onurcankaraman, 2022-02-24:
I didn't do anything different. I see Connected, but I can't see the data stream.

Today, when I want to send data to a different location and assets, I noticed that location and asset do not occur. Data flow is done only for test location and asset.

JeremyTheocharis, 2022-02-24:
You need to subscribe to the topic `ia/#` and not `#` as there is an ACL thats forbidding you from subscribing everything

onurcankaraman, 2022-02-25:
I'm using ia/# as a topic, but it doesn't change anything. (In the last case, I could see the data flow when I did this).
[image]

However, I restarted the pods with the thought, "Is there a problem with the pods?" and I encountered this error. I can't run any code right now.

[image]

I think you mentioned this in bug #837 but it is marked as resolved. How can I get this update?

Scarjit, 2022-02-25:
@onurcankaraman What version is your stack on ?

onurcankaraman, 2022-02-25:
0.8.0, if I'm looking in the right place  😄 
[image]

Scarjit, 2022-02-25:
Hm,

[auto plugin installation](https://github.com/united-manufacturing-hub/united-manufacturing-hub/blob/24776c3fc67e0581b107081127edf77291dc820d/deployment/united-manufacturing-hub/templates/nodered-statefulset.yaml#L46-L57) was tested in 0.8.0, you can still manually install them by opening the [Palette Manager](https://nodered.org/docs/user-guide/editor/palette/manager) and installing:
```
node-red-contrib-modbus
node-red-contrib-edge-trigger
node-red-contrib-opcua
node-red-contrib-s7
node-red-contrib-cron-plus
```

Is you stack a fresh install, or have you migrated from 0.7.X ?

onurcankaraman, 2022-02-28:
I did a clean install. I did not update. Depending on the situation, I will try the 0.8.x version on the virtual machine and continue with the 0.7.x version on the IPC. (For progress in the project. We can switch to 0.8.x again in the following stages.).

I'm trying to install manually.

JeremyTheocharis, 2022-03-02:
If the issue persists we could jump on a video call with you as we are interested in how you got to that point and solving it will be probably easier.

closed 2022-03-09 by onurcankaraman