united-manufacturing-hub/united-manufacturing-hub #864: Simulating a real-life event scenario

cvetkovski98, 2022-02-21:
Hi everyone,

I am looking into using and getting familiar with the United Manufacturing Hub. So far, I managed to set up and deploy the system on K3S and everything seems to be working fine for me as MQTT events emitted from the factory edge are written to Timescale and Redis. 

I am trying to simulate what a scenario of events would look like on the system and output some visualizations using Grafana. I went through the data model docs - [https://docs.umh.app/docs/concepts/mqtt/](https://docs.umh.app/docs/concepts/mqtt/) and I am somewhat familiar with the functionalities provided by the system.

Can someone please provide some guidance on how to approach modeling a sequence of events on a specific asset to resemble a real-life scenario?

Thanks!

JeremyTheocharis, 2022-02-21:
@oarne97 @Scarjit

oarne97, 2022-02-21:
Hi @cvetkovski98,

In example 2 (https://docs.umh.app/docs/getting-started/connecting-machines-creating-dashboards/) a Node-RED flow was created for a production line where the number of produced parts is measured using a light barrier.

It shows step by step how to read the sensor and assign it to the line. The data is then processed via MQTT and visualized with Grafana.

If there are any questions or points that are unclear within this documentation, please let me know and I will go into more detail.

I hope this could help you a bit and I'm looking forward to hear from you.

cvetkovski98, 2022-02-23:
Hi @oarne97, thanks a lot for the direction and the resources! I will be taking a look into that and update on the progress.

cvetkovski98, 2022-02-23:
Hi @oarne97, just wanted to update this with a quick thank you for the provided assistance. Instead of Node-RED, I went with a simple event publisher in Go [(gist here)](https://gist.github.com/cvetkovski98/247ca5395e9c653c020e5db3ddcc7fdf) that associates a product with a machine and goes through an order execution flow and item counting.

I found the docs quite easy to follow and the overall system was pretty straightforward to deploy on K3S thanks to the well-prepared helm charts. In the end, we found it easier deploying the edge and server on separate VMs compared to the combined Helm chart. I found some trouble visualizing things in Grafana as it only allowed me to see data from the default `testLocation`, and I couldn't seem to figure out how to visualize machines from locations added later. I attached my resulting dashboard below.

Overall I found the experience quite interesting and looking forward to exploring some more.

[image]

JeremyTheocharis, 2022-02-23:
Cool that you managed to get it working! Feedback is always appreciated. If you need any help with the dashboard, feel free to ask us. Maybe we can put you in touch with one of our technicians.

seanrmurphy, 2022-02-24:
Hi @JeremyTheocharis,

Thanks for this. I can share the following feedback on our experience:
- we did not really follow the installation instructions as they focused on K3OS initially - we just wanted to bring it up on VMs to see how it feels. We have pretty good experience with k8s and helm, so this was reasonable for us (and as @cvetkovski98 noted, your helm charts were good!). That said, it did take us a few days of looking at logs in k8s and troubleshooting to get the system up and running
- we did not really understand the model that you envisage generally but in particular the organization of MQTT subjects and how these map to data which is persisted - this is somehow presented here (hitttps://docs.umh.app/docs/concepts/mqtt/) but when starting this was prob too much of a deep dive for us
- when bringing the system up, it was not clear to us that we needed to present credentials to the MQTT broker (vernemq, I think) to send a message - not sure if this is documented - we found this from the acl configs in the container
- the model regarding factories/locations/assets etc and in particular how to work with multiple factories is not completely clear to us
- the functionality of the grafana plugin is not specifically clear - does this provide some pre-baked panels which you guys have or is this just to connect to one of the services? As a matter of interest, why not connect to timescale directly?

Anything to add @cvetkovski98 ?

Happy to have a chat with you on this if you'd be up for it!

Scarjit, 2022-02-24:
@seanrmurphy Thanks for the feedback.
To answer a few of your questions:

 - There are two grafana plugins we use.
   - The first provides access to factoryinsight.
     - This plugin provides some easy to use performance indicators
   - The second allows us to run multiple customers on the same instance, by acting as an authentification proxy

Regarding the documentation problems, be assured that we will make our documentation better.

JeremyTheocharis, 2022-02-24:
@seanrmurphy @cvetkovski98 

Thank you for the good feedback! This will help us when working on the documentation, especially when focusing on IT people (normally the system is used by mechanical engineers). Any idea where we could place your documentation suggestions?

If you have any other questions or need help, feel free to ask. Also if you use it in a scientific publication feel free to report back. We have an entire section in our documentation just related to scientific works: https://docs.umh.app/docs/publications/