united-manufacturing-hub/united-manufacturing-hub #867: Bug Report: adding new datasource not possible in UI

JeremyTheocharis, 2022-02-21:
### Expected Behavior

It should be possible to add a new datasource in Grafana.

### Current Behavior

The API Key needs to be in UUID format. Even then when pressing "save&test" in Grafana it results in a "502 Bad Gateway" error.

### Steps to Reproduce

1. Install the stack
2. Add a new customer in yaml with using 
```yaml
customers:
   testcustomer: testKey
3. create a new datasource and press "save&test"

### Context (Environment)

This is required to use a different MQTT topic instead of factoryinsight and to access the data.

### Version

v0.8.x

JeremyTheocharis, 2022-02-21:
only works when additionally (not exclusevly!!!) specifying

```
dashboards: {}
  dashboardsConfigMaps: {}
  datasources:
    datasources.yaml:
      apiVersion: 1
      datasources:
      - access: proxy
        editable: false
        isDefault: false
        jsonData:
          apiKey: $FACTORYINSIGHT_PASSWORD
          apiKeyConfigured: true
          customerId: $FACTORYINSIGHT_CUSTOMERID
          serverURL: http://united-manufacturing-hub-factoryinsight-service/
        name: umh-datasource
        orgId: 1
        type: umh-datasource
        url: http://united-manufacturing-hub-factoryinsight-service/
        version: 1
      - access: proxy
        editable: false
        isDefault: true
        jsonData:
          apiKey: ddddddddddddddddddddddd
          apiKeyConfigured: true
          customerId: ddddddddddddddd
          serverURL: http://united-manufacturing-hub-factoryinsight-service/
        name: umh-datasource-dddddddddddd
        orgId: 1
        type: umh-datasource
        url: http://united-manufacturing-hub-factoryinsight-service/
        version: 1
```

Scarjit, 2022-02-22:
@JeremyTheocharis I can't find the yaml key "customers", so i can't replicate this issue

JeremyTheocharis, 2022-02-22:
> @JeremyTheocharis I can't find the yaml key "customers", so i can't replicate this issue

You should replicate it even without. To fix it you can add this key, which is not in the yaml by default.

closed 2024-10-22 by JeremyTheocharis