united-manufacturing-hub/united-manufacturing-hub #888: Messages not persisted on postgres

edHiga, 2022-02-24:
### Expected Behavior

Messages from MQTT are persisted on postgresDB

### Current Behavior

Messages are generated on NodeRed and using the same configuration as usual (no changes on this system since 2 weeks), data is also visible through MQTT Explorer with the correct arragement (ia/factoryinsight) but is not persisted on postgresDB. Lens shows no pod errors

### Steps to Reproduce

System was performing fine, after the WAN connection drop it didn't have the messages persisted anymore

### Context (Environment)

No changes - system was running with the same configuration and parameters before the connection drop. We just want to keep as it was

### Version

not applicable

JeremyTheocharis, 2022-02-25:
I see that you have restarted VerneMQ. In this case you need to restart mqtt-to-posgtresql as well. Just delete the pod and it will restart.

JeremyTheocharis, 2022-02-25:
If this does not solve your issue, could you then post the logs of mqtt-to-postgresql?

JeremyTheocharis, 2022-03-02:
Can we consider your issue closed? Did you manage to fix it?

closed 2022-03-08 by JeremyTheocharis