united-manufacturing-hub/united-manufacturing-hub #903: Extension of Digital Shadow

RobertFeldmannMck, 2022-03-07:
[image]

**Is your feature request related to a problem? Please describe.**

In a non-digital state of our factory, we conduct manual quality control activities. As a part of this, a one-time analysis of the likelihood of failure of components and corresponding testing frequency was conducted, not accounting for improvements in the production process. In the digital state, the frequency of testing procedures for incoming, work-in-progress and finished products are dynamically adjusted based on long term production and quality data. Operators will receive automatic notifications on dashboards when testing is due. We are currently already using the digital shadow solution to track the asset through our manufacturing processes and for in-line visual quality inspection and safe everything in a product cv.

**Describe the solution you'd like**

I would like to have an addition to the data model of the asset to include a digital representation of the likelihood of failures (i.e., if the last 3 quality controls out of a 100 have failed, a 3% likelihood of failure should be saved). This figure should be dynamically adjusted based on both in-line and in-lab quality control results. 

Based on this likelihood of failure, the sampling frequency (i.e., 1 in 100 production parts) needs to be associated. Based on the sampling frequency, specific products needs to be tagged and marked inside the database to be brought to the lab for quality control.

Finally (already covered with the digital shadow implementation), the quality control activity is conducted in the quality laboratory and the result (quality is OK, quality is not OK) should be saved in the database and visualizable in a Grafana dashboard.

See exemplary dashboard attached:

[image]

[image]

JeremyTheocharis, 2022-03-08:
We did a time estimation and @kruegeralex will contact you with more information

Scarjit, 2023-03-31:
Is this still relevant @JeremyTheocharis  ?

closed 2024-10-22 by JeremyTheocharis