united-manufacturing-hub/united-manufacturing-hub #942: Bug Report: values not showing up in Grafana

JeremyTheocharis, 2022-03-30:
### Expected Behavior

The distinct processvalues should be shown in Grafana
[image]

### Current Behavior

There are cases when it does not show up at all. The content of the "value" field is still the same even after changing the asset.

Browser returns 502
[image]

The corresponding SQL query takes at least one minute
[image]

### Steps to Reproduce

1. Have a lot of processvalue entries for that specific asset
2. Open Grafana

### Context (Environment)

Customer in Atlanta

### Possible Solution

Datamodel rework as planned

### Version

v0.8.x

Scarjit, 2022-05-05:
I have tested adding another index to valuename, and seen speedups of 50-100 times (from 5-6s down to 50-200ms).
This might also be, because with the new index, timescale can apply [skip scan](https://www.timescale.com/blog/how-we-made-distinct-queries-up-to-8000x-faster-on-postgresql/)

JeremyTheocharis, 2022-05-09:
is this already in a newer version?

Scarjit, 2022-05-09:
No

Scarjit, 2022-05-09:
See #1076

Scarjit, 2022-06-03:
Closed: Fixed in [0.9.6](https://github.com/united-manufacturing-hub/united-manufacturing-hub/releases/tag/v0.9.3)

closed 2022-12-21 by Scarjit