united-manufacturing-hub/united-manufacturing-hub #992: Bug Report: upgrading from prerelease3 to prerelease4 not possible

JeremyTheocharis, 2022-04-22:
### Expected Behavior

I should be able to upgrade from prerelease3 to prerelease4

### Current Behavior

[image]

Upgrading fails

### Steps to Reproduce

Upgrade from prerelease3 to prerelease4

### Context (Environment)

Customre project

### Version

v0.9.x

JeremyTheocharis, 2022-04-22:
Solution: delete the statefulsets barcodereader, sensorconncet and nodered and then perform the upgrade. no data will be lost

JeremyTheocharis, 2022-05-09:
When using the server stack also deployments need to be deleted

Scarjit, 2022-06-03:
@Sphingobium This could be a topic for the new doc's ("How-to upgrade the stack")
There is a notion article, but it should be public information

Scarjit, 2022-12-21:
Documented

closed 2022-12-21 by Scarjit