united-manufacturing-hub/united-manufacturing-hub #1001: Bug Report: topicmap update requires deleting the statefulset

JeremyTheocharis, 2022-04-26:
### Expected Behavior

When changing the topicmap in the Helm chart, the pod should be restarted with the new environment variables

### Current Behavior

It is not restarted and the entire statefulset needs to be deleted and redeployed in order to get the new topimap

### Steps to Reproduce

1. CHange the topicmap in Helm Chart
2. See how it is not changed in the statefulset

### Context (Environment)

Venice project

### Version

v0.9.x

Scarjit, 2022-04-26:
Cannot reproduced, just updated values in 1-8 using the helm release, and it instantly updated

JeremyTheocharis, 2022-04-27:
Maybe then false alarm

closed 2022-04-27 by JeremyTheocharis